FreeCAD/FreeCAD #7585: compilation error

looooo, 2022-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

no forum discussion

### Version

0.21 (Development)

### Full version info

```shell
rev_number: 30610 (Git)
branch_name: master
commit_date: 2022/10/10 01:00:57
commit_hash: d8d9e21d238bd37a708d3ddab69730fe3f682429
remote_url: https://github.com/FreeCAD/FreeCAD master
```

### Issue description

```
$SRC_DIR/src/Mod/Fem/App/FemMesh.cpp:1174:21: error: unknown type name 'string'; did you mean 'std::string'?
        std::vector<string> token_results;
                    ^~~~~~
                    std::string
$BUILD_PREFIX/bin/../include/c++/v1/iosfwd:250:65: note: 'std::string' declared here
typedef basic_string<char, char_traits<char>, allocator<char> > string;
                                                                ^
$SRC_DIR/src/Mod/Fem/App/FemMesh.cpp:1228:21: error: unknown type name 'string'; did you mean 'std::string'?
        std::vector<string> token_results;
                    ^~~~~~
                    std::string
$BUILD_PREFIX/bin/../include/c++/v1/iosfwd:250:65: note: 'std::string' declared here
typedef basic_string<char, char_traits<char>, allocator<char> > string;
                                                                ^
$SRC_DIR/src/Mod/Fem/App/FemMesh.cpp:1298:21: error: unknown type name 'string'; did you mean 'std::string'?
        std::vector<string> token_results;
                    ^~~~~~
                    std::string
$BUILD_PREFIX/bin/../include/c++/v1/iosfwd:250:65: note: 'std::string' declared here
typedef basic_string<char, char_traits<char>, allocator<char> > string;
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

looooo, 2022-10-10:
I guess the issue was introduced with this commit:
https://github.com/FreeCAD/FreeCAD/commit/3b4cbbb9430368205ccd5b11cd29b44e2c7925f4

The (conda-based) weekly builds are not working anymore.
@donovaly can you help resolving this issue?

looooo, 2022-10-10:
using std::string for the mentioned cases, results in another issue:
```
$SRC_DIR/src/Mod/Fem/App/FemMeshShapeObject.cpp:85:23: error: member access into incomplete type 'SMESH_Mesh'
    newMesh.getSMesh()->ShapeToMesh(shape);
```

donovaly, 2022-10-10:
I could not reproduce and also the CI compiler showed no error, therefore I tried to fix the issue based on your error messages:
https://github.com/FreeCAD/FreeCAD/commit/a96a3a505f63a589cc6dad02478c4df135cf7fdd

Does this work for you?

If you find more issues, please report them directly to the developers forum because this usually attracts immediate actions. (I can currently only do something in the evenings.)

closed 2022-10-10 by donovaly

looooo, 2022-10-14:
@donovaly not sure if this is also related to the refactoring. Maybe @wwmayer can help fixing this compilation issue.

adrianinsaval, 2022-10-14:
this issue should be reopened if compilation is still failing @donovaly

donovaly, 2022-10-15:
> no still not working:

@looooo , please report such compilation errors directly to the developer's forum that they can quickly be fixed: https://forum.freecadweb.org/viewforum.php?f=10

> $SRC_DIR/src/Mod/Fem/App/HypothesisPy.cpp:60:12: fatal error: StdMeshers_TrianglePreference.hxx: No such file or directory
>    60 | #  include <StdMeshers_TrianglePreference.hxx>

This include has not been added recently. Since the compiler cannot find it, you use an old SMESH (older than version 7). But since it worked for you the last months, have you changed your build system and downgraded SMESH?
What is your SMESH version? SMESH 7 was released some years back therefore I would wonder if you have such an old version.

If you use SMESH >= 7, then the compiler cannot read the version information of SMESH.

donovaly, 2022-10-15:
> this issue should be reopened if compilation is still failing @donovaly

No, this is a new issue. To keep the overview please report every bug in its own issue. however, since it is a compilation issue, it should better be resolved the short way by posting into the developer's corner forum.

looooo, 2022-10-15:
Sorry, I don't have the resources to get into this troubles right now. In my eyes the issues are due to changes of the freecad code and is not related to packaging. If we can't solve this, the weekly builds are broken for now.

donovaly, 2022-10-15:
> Sorry, I don't have the resources to get into this troubles right now. In my eyes the issues are due to changes of the freecad code and is not related to packaging.

Let's focus to fix the issue. Our two CI compilers and Visual C with LibPack have no problems so we will find the cause of your issues.
As you can see in the Git log the code of the file your compiler misses was there since years. So FreeCAD was not changed.  Therefore we must find out what your compiler tries to load a files that should not be loaded at all.
Technically I see 2 possible reasons:
1. you either don't build with SMESH support or use a wrong SMESH package
2. the compiler cannot determine the SMESH version number

For case 2, can you please first checkout current master and try again. If this still fails, can you please apply this PR on top of master and try again: #7610

If this does not help you. I ask again, what version of SMESH do you use?

looooo, 2022-10-16:
it's working again (magic?):

https://dev.azure.com/sppedflyer/feedstock-builds/_build/results?buildId=564&view=results

donovaly, 2022-10-17:
> https://dev.azure.com/sppedflyer/feedstock-builds/_build/results?buildId=564&view=results

Awesome!

First now I understand. This was the fix:
https://github.com/FreeCAD/FreeCAD/commit/63cc0422991d1a797e40bf7d9b9a11661dfcb068

So the compiler was not aware of the SMESH version and therefore included an inexisting file.

So that we now have weekly builds again, can you please try to create builds for the toponaming branch? That would be very nice.

looooo, 2022-10-17:
thnaks @donovaly I am going to setup a branch to build the topo-naming branch.

luzpaz, 2022-10-17:
@looooo for uniformity, just use `toponaming` if you don't mind.

looooo, 2022-10-17:
@luzpaz too late

looooo, 2022-10-17:
here are the logs for the toponaming branch:
https://dev.azure.com/sppedflyer/feedstock-builds/_build/results?buildId=566&view=logs&jobId=5be07ae1-d8ba-5406-47b6-8e3a3a12f825&j=d0d954b5-f111-5dc4-4d76-03b6c9d0cf7e&t=841356e0-85bb-57d8-dbbc-852e683d1642

I do not have the resources currently to get into these issues...

commit a96a3a50 (2022-10-10): use explicit prefix std:: for vectors

- also add potentially missing include
- this commit is an attempt to fix #7585
commit 1fd8b8cd (2022-12-27): use explicit prefix std:: for vectors

- also add potentially missing include
- this commit is an attempt to fix #7585