FreeCAD/FreeCAD #7587: API function to access number of cols and rows in a spreadsheet

KeithSloan, 2022-10-10:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=50259

### Subproject(s) affected?

Other (specify in description)

### Idea description

Munther Hindi a co-developer of the GDML workbench has added functionality to import/export GDML Matrix properties to spreadsheets, his code needs to be able to determine the current number of cols and rows in a spreadsheet.
The number of cols/rows is not available via the API and he has had to resort to the work around as per the FreeCAD forum discussion, which it has to be said is a real Kludge of a solution. Please can we have a API facility to access the current number of cols and rows in a spreadsheet.

Thanks  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-11-04:
When you say you need the "current number of rows and columns" do you mean you want to know the largest row/column that has data in it? Or a count of the number of rows/columns that have data? What about providing a wrapper around the C++ `getNonEmptyCells()` function? Would that get the needed result?

KeithSloan, 2022-11-04:
Checking with Munther

Hi Keith,

Yes, Max row/column with data. 

Thanks,

Munther

On Thu, Nov 3, 2022, 10:24 PM Keith Sloan <keith@sloan-home.co.uk> wrote:

    Munther

    I assume what is required is the largest row/column with data, want to check with you before I respond

    Thanks Keith
    
    Also if you check the thread others have ask for this in the past.

chennes, 2022-11-04:
I think it would be a pretty fast calculation to make in the C++ code: I wouldn't expect the function to be called that often, so we can likely just brute-force it for the time being. It's probably really a set of four functions:
```
int getLastUsedRow();
int getLastUsedColumn();
int getLastNonEmptyRow();
int getLastNonEmptyColumn();
```

KeithSloan, 2022-11-04:
Having to resort to

````
def processSpreadsheetMatrix(sheet):
    # Stupid way of finding how many rows. columns:
    # increase col, row until we get an exception for that cell
    # You would think the API would provide a simple function
    def ncols():
        n = 0
        try:
            # TODO: deal with case n > 26
            while n < 26 * 26:
                sheet.get(chr(ord("A") + n) + "1")
                n += 1
        except:
            pass
        return n

    def nrows():
        n = 0
        try:
            while n < 256 * 256:
                sheet.get("A" + str(n + 1))
                n += 1
        except:
            pass
        return n

print("add matrix to define")
    coldim = ncols()
    rows = nrows()

    ....
````

This is being called for every defined Matrix, so potentially a lot, so they are being called often. 

To quote a wmayer forum post
````
Extending
This is so to speak the opposite of embedding. When you have working a C++ function or class then you can expose it to Python so that in Python code you can invoke the C++ code.
There exist many ways how to expose C/C++ to Python. There is e.g. the wrapper generator framework called SWIG, the boost.python library, the pybind11 template library, also PyCXX or you can do everything directly with Python's C API.
````
Would it really be such a hard thing to add bindings for these functions?

chennes, 2022-11-04:
> Would it really be such a hard thing to add bindings for these functions?

No, not at all, sorry I wasn't clear. It's that there _is no function_ to add a binding to at all right now: even the C++ doesn't provide simple access to "what is the highest row used?" and "what is the highest column used?" So I'm proposing that we write the aforementioned four functions in C++ (which will be fast), and then write Python bindings to those (which will be easy). If they are going to be called a lot, we may need to be more clever than just a brute-force approach, though. Even the C++ will slow down on a large enough data set.

KeithSloan, 2022-11-04:
Thanks for making things clearer. I think python bindings for new C++ functions would be a good approach.

chennes, 2022-11-04:
OK, have a look at PR #7697, which I think will provide what you need. I collapsed the functions to provide both the upper and lower bounds of the used range in a single function call.

KeithSloan, 2022-11-10:
Munther has looked at the code and here are his comments and I think we are good to go.

Hi Keith,

I looked at the diffs on github. There seems to be two sets of routines, one set returns "used cells" and the other returns "non empty cells". It is not clear to me what a used but empty cell is. Perhaps the documentation can make that clear.

Also, the documentation for the getNonEmptyRange() method says
Note that the actual first and last cell
of the block do not necessarily contain anything.

At first that took me aback (how could a range of nonempty cells have empty cells at the boundary?). But upon reflection I understand how that can happen:

x x x x
x x x
x x x   <-- cell in max range is empty; it is in max range because it's row is max row)

The Python wrappers seems to cover what we need. Of course, I have no idea if the underlying C++ code is bug free or not. That is up to them.

Munther

chennes, 2022-11-10:
Excellent, thanks for the comments. I'll get this merged.

closed 2022-11-10 by chennes

chennes, 2022-11-10:
Also, @KeithSloan, you can tell Munther that the difference between "used" and "non-empty" is that a cell is "used" if it has either data or formatting in it, but is only non-empty if it has data.

PR #7697 (2022-11-10): Spreadsheet: Add Python API for getUsedCells
Also adds access to `getNonEmptyCells()`, and unit tests for both. In a second commit, two new C++ functions are introduced to get the range of non-empty and used cells, `getNonEmptyRange()` and `getUsedRange()`. These functions return a pair of cell addresses representing the bounding box of the data in the sheet. Python wrappers and unit tests are also provided.

Designed to fix #7587.
commit 11dd7fc5 (2022-11-10): Spreadsheet: Add Python API for getUsedCells

Also adds access to getNonEmptyCells, and unit tests for both. Designed
to fix #7587.