FreeCAD/FreeCAD #7589: [Bug] Formula Editor not working properly in sketch

funny0facer, 2022-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72400

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

When I enter a formular into the formular editor for a specific sketch constraint, it calculates correctly, but does not accept the value. 
It seems to have a connection to the constraint I use in the sketcher attachment position. 
See gif:
[image]
FreeCAD file is attached.

### Anything else?

[FormularEditorBug.zip](https://github.com/FreeCAD/FreeCAD/files/9748293/FormularEditorBug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-10-13:
Gist: https://forum.freecadweb.org/viewtopic.php?p=632239#p632239

Zolko-123, 2022-10-28:
Not enough information. Please describe the problem better, and ideally nail it down to the very core of the problem.

luzpaz, 2022-10-29:
@Zolko-123 https://github.com/FreeCAD/FreeCAD/issues/7589#issuecomment-1277511911

funny0facer, 2022-11-01:
from my point of view, the worst about this behaviour is, that there is no warning/error or something else. You can clearly see, that the result of 5/1 does not match the shown result (2mm).
If there would be an error message (or something else), the user could try to fix the problem in the sketch. Otherwise, the user might never identify that changing a value does not actually affect the sketch.

AjinkyaDahale, 2023-04-14:
Does this bug still exist after merge of #8311?

funny0facer, 2023-04-15:
seems fixed on:
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32821 (Git) AppImage
Build type: Release
Branch: master
Hash: eaf9d43f37238c33dd1c47b859a6fdf5b7f629fc
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]

thanks! Bug can be closed from my point of view.

closed 2023-04-15 by funny0facer