FreeCAD/FreeCAD #7596: [Bug] Links break when opening a Symlinked FCStd

leoheck, 2022-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

(tbd)

### Version

0.21 (Development)

### Full version info

```shell
latest AppImage (tbd)
```

### Issue description

When a Symlinked file is opened with Freecad, if it is not in the same folder with it's target (which is most of the time the case) the links to the linked files inside the file doesn't work anymore, since they are not relative to the path of the symlink.

A simple solution would follow the symbolic link to open the actual file, instead, when the given file is a symlink.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2022-10-11:
> A simple solution would follow the symbolic link to open the actual file, instead, when the given file is a symlink.

And maybe, iterate since multi-symlink levels may exist.

luzpaz, 2022-10-15:
I seem to recall an pre-existing issue regarding symlinks....brb

Zolko-123, 2022-10-24:
This is not a [Bug] but the very nature of symlinks. If the user needs the linked (as per App::Link I guess) file to open correctly he has to symlink the linked file into the same folder.

This issue should be closed

leoheck, 2022-10-24:
No, this is a bug. When you symlink a file, and since Freecad does not adjust paths internally, Freecad should open the target instead.

luzpaz, 2023-04-13:
@FlachyJoe care to weigh-in ?

luzpaz, 2023-09-27:
CC @FlachyJoe

FlachyJoe, 2023-09-27:
@luzpaz this issue is about App::Links in a symlinked file. I can't see relation with save problem in junction.
Also, I have no knowledge about App::Link, so I can't do anything for this problem.

luzpaz, 2023-09-28:
@leoheck just out of curiousity, does realthunder's branch address this issue ?

leoheck, 2023-09-28:
Sorry @luzpaz I have never used this branch. Unfortunately, I can't answer this.

luzpaz, 2023-09-28:
@realthunder do you have a moment to weigh-in on this ? (specifically since you were involved with App:Link code)

adrianinsaval, 2023-10-02:
I tend to agree with zolko that this does not seem like a bug.

luzpaz, 2023-10-15:
Anyone able to test realthunder's PR ?

maxwxyz, 2024-08-28:
Is this then still relevant if not a bug?

leoheck, 2024-08-28:
I think this is still relevant. When you symlimk something, you expect this to keep working. But the internal structure of links break, since they are not using the right path anymore. But I will check if I can make an example to show this.

adrianinsaval, 2024-08-28:
> Is this then still relevant if not a bug?

I guess it could be a feature request to optionally follow a symlink target before opening but I'm not sure if this should be default behavior

maxwxyz, 2024-08-29:
I'll close this then as currently not planned, if it works as expected and there is no compelling usecase. I'm not sure who would go ahead and solve this issue if we leave it open.
Let me know otherwise, we can reopen this anytime.

leoheck, 2024-08-29:
This is not the right behavior. The parts linked inside the file should refer to the original path, and not to the path of the link, and this is a really simple fix, otherwise the linked design won't work as expected because linked files won't be found anymore.

luzpaz, 2024-08-29:
> But I will check if I can make an example to show this.  

@leoheck when you can make said example, please feel free to reopen.

leoheck, 2024-08-29:
@luzpaz cool, however I cannot reopen.

[image]

luzpaz, 2024-08-29:
@leoheck that's weird since you authored the ticket. Not sure why. But nonetheless, if you can follow-up someone will re-open

leoheck, 2024-08-29:
Yeah, I am seeing this since a long time.

leoheck, 2024-08-29:
I put this example here.
[image]

This happens when I open the link
[image]

This is the symlinked file. The main file. 
[image]

This is the design freecad-linked on the main file.
[image]

Here is the example, it should work in Linux, MaOS (and nowadays on Windows I guess if something is enabled, at least it works on WLS)

[broken_links.zip](https://github.com/user-attachments/files/16805669/broken_links.zip)

Zolko-123, 2024-08-30:
@leoheck : the file doesn't open : 

```
Traceback (most recent call last):
File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file /home/hubertz/Tmp/broken_links/main_file_link.FCStd: basic_ios::clear: iostream error
```

Zolko-123, 2024-08-30:
I'm using Linux (Debian) and from the command line I have a different error (tested with 3 FreeCAD versions) : 

`Unknown exception while processing file: main_file_link.FCStd`

leoheck, 2024-08-30:
This is weird. Did you check if the link is broken?
You can do that with this: `file main_file_link.FCStd`

leoheck, 2024-08-30:
> tested with 3 FreeCAD versions

Did you test with the AppImage too, @Zolko-123 ?

Zolko-123, 2024-08-30:
I think that the ZIP you provided is corrupted, the file is only 38 bytes size. Did you make tests after downloading your own file ?

leoheck, 2024-08-31:
Yes, when I answered this https://github.com/FreeCAD/FreeCAD/issues/7596#issuecomment-2321142502, I downloaded the zip from here. But I will check it again just in case.

leoheck, 2024-08-31:
I am using the following Freecad version now: `FreeCAD_weekly-builds-38555-conda-Linux-x86_64-py311.AppImage`

Then I created a new example (the first one was not bad)
[broken_links_2.zip](https://github.com/user-attachments/files/16823152/broken_links_2.zip)

These are the files on this second example:

```bash
~/broken_links_2 …
➜ tree
.
├── assembly_link.FCStd -> project/assembly.FCStd
├── broken_link.FCStd -> project/deleted_file.FCStd
└── project
    ├── assembly.FCStd
    ├── assembly_link.FCStd -> assembly.FCStd
    └── cube.FCStd

2 directories, 5 files
```

1. This is the output of launching the (working) symlink from the command line. The link inside Freecad is broken, since `assembly.FCStd` linked by `assembly_link.FCStd` expects `cube.FCStd` at the same directory.

```bash
➜ freecad assembly_link.FCStd    
[FREECAD_HEADERS_WERE_REMOVED]
<Exception> File '/home/lheck/broken_links_2/cube.FCStd' does not exist!
Exception opening file: /home/lheck/broken_links_2/cube.FCStd [File '/home/lheck/broken_links_2/cube.FCStd' does not exist!]
5.3e-08 <App> Document.cpp(2217): assembly_link#Link.LinkedObject: Link not restored
Linked object: Box
Linked file: cube.FCStd
```

2. This is the output when I double click the  `assembly_link.FCStd` to launch Freecad from the GUI. This is also the same thing as I launch it through the `File > Open` menu.

```
[FREECAD_HEADERS_WERE_REMOVED]
23:14:05  <Exception> File '/home/lheck/broken_links_2/cube.FCStd' does not exist!
23:14:05  Exception opening file: /home/lheck/broken_links_2/cube.FCStd [File '/home/lheck/broken_links_2/cube.FCStd' does not exist!]
23:14:05  3.7e-08 <App> Document.cpp(2217): assembly_link#Link.LinkedObject: Link not restored
Linked object: Box
Linked file: cube.FCStd
```

Notice that in this new example, I also have another link inside the project folder (`project/assembly_link.FCStd`) that works. This happens since it is in the same place where the expected Freecad Linked-Object is. 

This was also tested by downloading the `broken_links_2.zip` while I was editing this response.

These two examples should work if Freecad could open the symlink target _from its current path_; OR, when dealing with Linked-Objects, if the file is a symlink, they should look for the Linked-Objects relatively to the path of the symlink target.

Zolko-123, 2024-09-01:
Same `Unknown exception while processing file: assembly_link.FCStd` error. I think that the symbolic links are not preserved (or restored) by the ZIP. Can you try with tar.gz ?
 
 

[image]

leoheck, 2024-09-01:
I tried yesterday, Github does not support it. unless I make a tar.gz and then a zip.

But I downloaded the zip file yesterday, twice. You may have something broken at your and.

But it doesn't matter. You can replicate the example easily.

Create the main folder, call it `broken_freecad_links`
Create another folder inside of it, call it `project`.
Inside the project, create 2 Freecad files. One you can name it `cube.FCstd` and then add a Part > Cube inside.   The second one, name it `assembly.FCStd` and make a Freecad link to the Part Cube from the `Cube.FCStd`. 
After, out of the `project` folder... make a symbolic link named `something.FCStd` that points to the `assembly.FCStd`. Then launch `something.FCStd` with the Freecad and check it the Cube appears.

Zolko-123, 2024-09-01:
Ah ok I understand your point now. 

But let's take a different view on this: imagine I have a Library directory somewhere, and I symlink it into the project folder. I then include a part from that library into the assembly : what would happen ? And if that part is actually a sub-assembly with other parts – from the library – included there, what would happen ?

leoheck, 2024-09-01:
> In other words : I don't see any reason why one would insert (or open) a symlinked file

Symlinks exist to make a link to another file/folder. They are similar to Window shortcuts. I use them everytime. The operating systems support it and because of that Freecad shouldn't break when using them.

> , but I can see reasons for inserting parts from a symlinked directory

This is exactly the same. You should be able to use as a library a symlinked directory. I just didn't try to exercise this behavior yet.

leoheck, 2024-09-08:
> but I can see reasons for inserting parts from a symlinked directory.

I was experimenting this. 

It works when you add parts from a symlinked directory however it complains when saving, at least it complained to me. My setup was with the symlink and the symlinked directory at the same place as follows.

[image]

This is the complain when saving:

```
<PropertyLinks> PropertyLinks.cpp(3141): reloading partial document '/home/lheck/broken_links_lib/lib/part.FCStd' due to object Box
<App> Application.cpp(725): Identical physical path '/home/lheck/broken_links_lib/lib_a/part.FCStd'
  for file 'lib/part.FCStd'
  with existing document 'part' in path: '/home/lheck/broken_links_lib/lib/part.FCStd'
```

Now, changing the link to another linked folder, even if you have the same parts, with the same name, does not work since the links refer objects by they name, internally, on freecad. And even if you have the same object on each linked file, freecad link breaks when you change the symlink to another library. In this case, the link is removed/cleaned, inside the freecad. 

Freecad links should work like OS links since they are linking files.
Except that they have extra information which is the Freecd Object that it being linked.
So, there is the file path that should be respected using the real directory of the linked part.
And there is the freecad object name as a property inside the freecad that should remain even if the path of the file (or the path of the linked file) is broken.

Zolko-123, 2024-09-09:
```
Identical physical path '/home/lheck/broken_links_lib/lib_a/part.FCStd' for file 'lib/part.FCStd'
  with existing document 'part' in path: '/home/lheck/broken_links_lib/lib/part.FCStd'
```

that's interesting: so it recognizes internally that the linked and the real files are identical

leoheck, 2024-09-09:
yeah, as I could quickly see in the codebase, it has some places where symlink target are checked/used takes into account. 
i tried to modify something quickly but I could not make it build yet.

adrianinsaval, 2024-09-20:
> They are similar to Window shortcuts.

not really, symlinks exist on windows too and teh equivalent to shortcuts on linux are .desktop files, not symlinks.

@FreeCAD/cad-advisory-group do you see any value in supporting this workflow?

leoheck, 2024-09-20:
Its a simple thing. Instead of opening the link, it should open the reference to it. Nothing more. This is one line of code won't break anything.

Now, regarding to a symlinked folder, it should work and it kind of does. But if the symlink is broken the link inside the Freecad should keep the property of the linked target. The value of this property should not be removed... as far as I remember Freecad is cleaning the value of this property.

pierreporte, 2024-09-20:
I don’t fully understand the use case. I’ve never heard of using symlinks or similar things to work with a component library. When working without a PDM, these libraries are stored in folder near design files and linked normally. Usually everything is stored on a remote drive so that the team can work together. They are not copied across folders.

Breaking the behavior of symlinks sounds dangerous but I’m not an expert.

It would be great if there was a minimalist PDM for FreeCAD to allow users leveraging one of the greatest powers of such a system, which is the part database.

adrianinsaval, 2024-09-20:
> Its a simple thing. Instead of opening the link, it should open the reference to it. Nothing more. This is one line of code won't break anything.

it does that, but it treats it as if it where in the symlink's location, which is how most software that I know does it. Can you poin tme to one example of a software that opens the target and not the symlink? IMO you are trying to cram a functionality where it doesn't belong.

adrianinsaval, 2024-09-20:
I guess it can be useful when a part that you want to include in your assembly also includes other files which you don't want to bring into your directory. In that case you symlink only the parent file and freecad could fine the children in the target. Not a very common situation IMO.

In any case, you can ask realthunder to update and reopen his PR if this feature is desired

leoheck, 2024-09-21:
> it does that, but it treats it as if it where in the symlink's location, 

Sure, but the problem arises since Freecad does also have links inside, so then this is a different story. Because then freecad links wont work anymore.

sslupsky, 2024-11-30:
Iff I change the file name of a linked object it should be possible to update the link.  For instance, open two files ("Part A" and "Part B", create a part in both.  Add a binder in one of the parts ("Part B")  that references the part from the other file ("Part A").  Save them.  Now change the file name of the part that was used for the binder ("Part A" to "Part A1").  I encounter this error when I do so:
```
Exception opening file: /Users/stevenslupsky/Library/Mobile Documents/com~apple~CloudDocs/3d modeling/coro props/trees/Part A.FCStd [File '/Users/stevenslupsky/Library/Mobile Documents/com~apple~CloudDocs/3d modeling/coro props/trees/Part A.FCStd' does not exist!]
```

Presently, it does not appear possible to fix this without hacking the XML file.

[image]

Zolko-123, 2024-12-01:
> Presently, it does not appear possible to fix this without hacking the XML file

of course it is possible : all you have to do is to select the broken link and click on the button "Insert Part". This will recognize the broken link and will give you the possibility to reassign the target object. 

You must have Assembly4 installed for this, obviously. 

https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/Resources/icons/Link_Part.svg

FEA-eng, 2025-09-27:
I'm on Windows now and not familiar with Symlinks. @leoheck Can you confirm if it's still relevant?