FreeCAD/FreeCAD #7606: display problem with nested parts

epiray, 2022-10-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=72556

### Version

0.20 (Release)

### Full version info

```shell
has already been checked to be present in previous versions 0.18 and 0.19, see forum thread. I also see it in the daily build versions.
```

### Issue description

From the forum thread:

In complex designs, I use nested parts for assembly. Once I nest more than 29 parts, they no longer display correctly. Either they are not transformed any more, or the wire frame stays put, while the volume shading of the assembly still transforms. See the following example:

nested_parts_bug.png
    nested_parts_bug.png (992.26 KiB) Viewed 172 times

I have boiled the problem down to the following example, in which each nesting level contains a bar, and each level transforms upwards and 10° around the z axis:

nested_parts_bug_example.png
    nested_parts_bug_example.png (595.23 KiB) Viewed 172 times

level_029 still displays correctly, the following levels display on top of level_029. The geometry logic still seems fine, when I blindly click at a location where I expect the correctly transformed part to be, it selects o.k.

Could not attach the example FCStd file, please refer to the forum thread.

### Anything else?

would be nice if I could attach FCStd files here.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-10-15:
Added screenshots + file attachment + About info  
to the OP

maxwxyz, 2024-03-09:
I've ran into this issue myself many times, often with assemblies.
When trying to move over commercial companies to use FreeCAD, their complex PLM setups and file/assembly structures cannot be displayed with FreeCAD as the tree currently looks like. The indentation of the sub parts or features makes it very hard to navigate and find the right info. 
In 0.22dev the tree can be overlayed with transparency over the model but the text is still cut off.
@obelisk79 maybe something for the DWG?

maxwxyz, 2024-03-09:
To add to this from the FreeCAD Day 2024 Complaint Session:
**The tree isn’t actually just a tree its a collection of things used to construct it**

obelisk79, 2024-03-10:
Indeed the treeview has some issues with information management. But in reading the original bug description here, it seems what's being described is that the 3d view has display issues after so many levels of nesting within the tree structure.

In relation to the reported issue, it may be a problem which stems from using Coin3D as a rendering engine as it was reported in the linked forum thread to not be present in linkstage which has implemented a different renderer.

epiray, 2024-03-10:
This remains a significant problem for us. To maybe give this a nudge, let me announce that epiray is offering a donation of EUR 500 once this bug is fixed.

maxwxyz, 2024-09-01:
@kadet1090 this should also be considered with https://github.com/FreeCAD/FreeCAD/issues/16152 and https://github.com/FreeCAD/FreeCAD/issues/16155

benj5378, 2024-10-19:
@epiray, are the 500$ still in play?

epiray, 2024-10-20:
Yes, they are. We are working with the conda-Linux-x86_64-py311.AppImage versions, this is where we need it fixed first.
We do complex designs, and frequently hit this limit.
Best,
Wolfgang

mwganson, 2024-10-22:
This script can create a file to illustrate the issue.

```
count = 50
doc = FreeCAD.ActiveDocument if FreeCAD.ActiveDocument else FreeCAD.newDocument()
previousPart = None
for i in range(count):
    part = doc.addObject("App::Part","Part")
    part.Placement.Base = FreeCAD.Vector(0,0,1)
    box = doc.addObject("Part::Box","Cube")
    box.Length = i if i else 0.1
    part.Group = part.Group + [box]
    if previousPart:
        previousPart.Group = previousPart.Group + [part]
    previousPart = part

```

50 App::Part objects are created, each nested inside the previous part.  Each part gets a Placement of Base = (0,0,1).  50 cubes are created, each cube gets a Length of i where i ranges from 0 to 49, the first getting 0.1 since 0 length is invalid.

[image]

Expectation is a triangle up until the last cube, which is what we get up to the first 30 parts, but the remaing 20 all get rendered at the same placement as Part029 (containing Cube029).

Curiously, how many get rendered can depend on the zoom level.  Closer you zoom in the fewer that get rendered:
[image]

Another curiosity is when you hover the mouse pointer over one of incorrectly rendered cubes, the preview / preselection shown in the 3d view is in the expected location:
[image]

Selecting in apparently empty 3d space will select the cube that is invisibly there. 

Select "Part" and create a mesh from it in Mesh workbench and you get a mesh that looks like you would expect but for the bug:
[image]

wwmayer, 2024-11-11:
See https://github.com/FreeCAD/FreeCAD/pull/17800