FreeCAD/FreeCAD #7623: [Bug] <Default theme has poor usability>

sliptonic, 2022-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=72672

### Version

0.21 (Development)

### Full version info

```shell
Affects all versions

[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.30623 +33 (Git)
Build type: Unknown
Branch: (HEAD detached at upstream/pr/7440)
Hash: c74a0ad3e55cd853668c6ae183a05daa2ec5df16
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.52
  * FeedsAndSpeeds
  * ExtremeProDark 1.0.4
  * FeedsAndSpeeds.bak
  * Dracula 0.0.2
[/code]
```

### Issue description

FreeCAD now has theme packs that allow users to change the appearance of the application easily and consistently.  This is a great feature to allow customization and to meet a wide variety of use-cases. 

The first experience with FreeCAD that new users have, however, is the default theme.  This default, including background gradient and color selection in several key workbenches has problems that adversely affect adoption, learning curve, and usability.  A new default theme should be created that meets at least the following four objectives.

1. It should be broadly usable in all default workbenches.  
2. It should be usable by a wide range of users including those with the most common forms of color-blindness
3. It should reproduce reasonably well in grayscale images so that print documentation accurately reflects the FreeCAD interface.
4. It should be attractive and inviting.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2022-10-20:
The existing default theme will continue to be included with future versions as "FreeCAD Classic"

chennes, 2022-10-20:
I figured "Classic" would mean "go back to the old way," so the current "FreeCAD Classic" preference pack sets it to "No stylesheet" and all the original colors. What exactly that looks like will then depend on your system setup, just like it does out of the box right now.

adrianinsaval, 2022-10-22:
another issue with the default colors: https://forum.freecadweb.org/viewtopic.php?p=635039#p635039
[image]
I want to make a PR to change these defaults and while almost anything would be better than the current defaults it would be nice to know if for the defaults we should prioritize dark themes or light themes, IMO we should make a poll to decide if our default theme should be dark or light so we can begin work on this type of things.
Of course the new default should be at least readable on both variants, but the choice of a darker or lighter color tone depends on what would be our default. The colors I want to propose are readable on both but IMO they don't look as good in the light theme as in the dark theme:
[image]

chennes, 2022-10-22:
Just as a point of technical reference, Apple's Human Interface Guidelines go into quite a lot of detail about how to ensure usability and accessibility (in a generic sense) for both light and dark color schemes: https://developer.apple.com/design/human-interface-guidelines/foundations/accessibility. They include guidance for both average and vision-impaired users. Obviously there is a lot of Apple-specific implementation detail in there, but they've clearly done some research into this area.

Glitchy-Tozier, 2022-10-25:
> Of course the new default should be at least readable on both variants, but the choice of a darker or lighter color tone depends on what would be our default.

There probably will never be a great solution if we stick with one default.
It seems more productive to use a system that predefines colors for each theme individually.

MisterMakerNL, 2022-10-25:
> > Of course the new default should be at least readable on both variants, but the choice of a darker or lighter color tone depends on what would be our default.
> 
> There probably will never be a great solution if we stick with one default. It seems more productive to use a system that predefines colors for each theme individually (= which combines highlights & themes).

You are right, I also tried a color that is readable on white and black text. But it breaks already when the text is gray. And it's just a tiny improvement not really worth doing it.
Best fix is too link stylesheets too preference packs so we don't get in situations like this.

adrianinsaval, 2022-10-25:
I'm not saying we should stick with a single default but that is what is technically feasible right now, IMO it is better to solve that now and then improve on it when the other changes being discussed are implemented.
IMO it is wrong to say "a better solution will come soon<sup>TM</sup>, so we should do nothing now" because it may very well be that the wait would be eternal...

MisterMakerNL, 2022-10-25:
> I haven't tried this again in a couple years but it's useful for evaluating colors:
> 
> https://forum.freecadweb.org/viewtopic.php?t=2914#p22402

Gimp won't run nice on my windows machine but I thought well there is probably a website for that...And off-course there is.
https://www.color-blindness.com/coblis-color-blindness-simulator/
And soon I realized the icons are the real issue here. Although I also have a solution for that but it's still requires a lot of work to finish the icon theming.

obelisk79, 2022-10-28:
First off, if the intent is for only a single set of default colors, we should first discuss and decide on a light vs dark theme before going to far down the rabbit hole with colors, and color theory and analysis of the current state of such in FreeCAD.

obelisk79, 2022-10-28:
An interesting albeit slightly biased analysis in favor of a dark theme:
https://www.venturus.org.br/en/darkmode-and-the-benefits-for-the-user-experience/

Multiple sources also indicate that dark UI trends dominate in preference polls with ranges between 85%-92%. That's actually a lot higher than I expected to see.

obelisk79, 2022-10-28:
In terms of tools for designing around contrast, color and vision impairment a simple tool/test one can do is take your sample colors and load them into image editor of choice. Photoshop, gimp or Pixlr and then completely desaturate the image. If there is still a high contrast ratio then you are good. If not, further adjustment is necessary.

obelisk79, 2022-10-28:
@MisterMakerNL in response to your comment regarding the icons, I would agree. I believe that contrast and usability particularly across a broad audience (ie color impaired) could explain why you see simplified and flat icons in a lot of professional software across multiple industries. I believe a good example of simple yet effective can be seen here:
https://forum.freecadweb.org/viewtopic.php?f=34&t=46633&start=40

st_phan's composition on the sketcher icons is phenomenal. They convey the same information more clearly, with less colors and they scale extremely well even to pretty small sizes. The analysis and process comments by him at the end of the thread are also highly valuable imo.

MisterMakerNL, 2022-10-28:
I have my own icons, and I really like too work with them too. But the project needs some more work.
https://github.com/MisterMakerNL/Linkstage_icons

Glitchy-Tozier, 2022-10-28:
Regarding light & dark themes: You probably could make a poll on the forum asking "Do you use light or dark theme?", to at least get _some_ hint about what FreeCAD's users prefer atm.

Zolko-123, 2022-11-01:
> Light Theme Pros: 

- with a shadow around windows, it's easy to see which window has focus (is on top)
- with a shadow around windows, it's easy to see where windows borders are, and to distinguish between window borders and widget panels
- black-on-white is a natural combination also used on paper

> Dark Theme Cons:

- with overlapping windows, there is no way to know where a window's borders are, to distinguish between a window border and window panels
- it's very difficult to know which windows has focus, therefore the designers are increasingly going towards full-screen apps which takes us back 30 years

Chrismettal, 2022-12-04:
As I don't feel that a consensus can be reached in deciding the default theme to be light vs dark, how about we collect both, a dark and a light default color set, and ask the user during onboarding to choose either one? 
Yes, that duplicates some of the work maintaining the core theme, but I think that this could be achieved with manageable effort.

I also agree that light themes seem to be used more in professional office environments, and dark themes are often used in darker at-home scenarios (both with valid reasons behind them) and as FreeCAD aims to be both, for hobbyists and professionals I think that light/dark onboarding could be a valueable inclusion.

Also @Zolko-123 what makes you think that dark themes can't have shadows and borders in 4 separate points?

Mikhail22, 2022-12-12:
Question: Should comments better be here or on the linked forum thread?
I'll put here few thoughs. 

First thought: there is no much expedience to speak about theme(s) before most elements are theme-able.
At first glance currently:
1. There is no any differentiation of line thickness and styles in Sketcher (thin, dotted, dashed, dash-dotted are needed). Probably similar issue for other WBs.
Custom line thickness/styles is crucial for improving UX, actually sometimes even more important 
than colors. Additional vertex styles should be available too.
(shoul it be a separate feature request?)

2. Some colors are hard-coded as well: Axes colors, selected item hover color.

Many other elements I guess too, just named what I noticed and wanted to tune first when I started 
to work. So regardless of light or dark choice, not much can be done with those.

Background : I prefer textured background to reduce eye strain. (another feature request I guess).
Otherwise solid color or slight radial gradient. 

My view on light/dark choice:
Light:
- I beleive it should be the choice for normally lit environment since it is physiologically predefined (dark 
objects and features on light background is how information is read by daylight creatures in general).
- It is de-facto standard for most systems in general.  

Dark:
- Looks kinda cool for some people.
- Might be better for poorly lit environment or in case of some rare eye diseases.

It is better to have both light and dark themes available. But again, see my first point.
So even if I was going to create light theme I just can't tune some colors, e.g. Sketcher axes colors can't be changed and those were obviously tuned initialy to match the dark bg.

life5ign, 2022-12-27:
This also happens for editing Annotations in TechDraw: 
[image]
https://github.com/FreeCAD/FreeCAD/issues/8093

Intuos5, 2023-03-02:
Question: will the default stylesheets be updated to remove legacy fixes for Qt bugs? There are quite a lot of these things that make it difficult to, for instance style Qt classes within other Qt classes.

adrianinsaval, 2023-03-02:
Somebody has to make a PR for it but if it's done it'll likely get merged

MisterMakerNL, 2023-03-03:
I don't even know what you mean by this, are these work arounds for old Qt bugs?

Intuos5, 2023-03-03:
> I don't even know what you mean by this, are these work arounds for old Qt bugs?

We discussed this esrlier thst there are various resets, resize to 0 and such to redolve Qt bugs. I don't know much about Qt, so it wpuld be a manual delete and see how it goes check for a default theme. That or have @Mistermaker's fixes and backport the default themes to his template.

MisterMakerNL, 2023-03-03:
I also started with the default ProDark theme files, but man that was annoying. I even dumped a lot of work into it, but I still got so frustrated that it took so much time figuring out what went wrong that I dumped it and started from scratch. (which was easier anyway :) )
I am just going to make some different versions of my theme file, maybe I can utilize Qpalletes.

Intuos5, 2023-03-04:
I will one day try to update my theme on your latest template. Maybe we could have a chat then, seeing that you are learning Qt to talk about these things.

tcurdt, 2023-05-20:
Color and contrast will always have a personal bias.
But I given all the research and guidelines out there it seems rather obvious this need some love in FreeCAD.

I personally just cannot use the classic/default theme - and I am not joking.
Especially the Sketcher WB is almost unusable for me.

I am happy to provide more feedback if someone wants to take a stab at a better theme.
Or at least there is a willingness to fix this usability problem.

MisterMakerNL, 2023-05-20:
@tcurdt I actually spend some time on updating the old theme's to my updated theme layout. It has a ton of fixes and also removes a ton of fixes, and it optimizes screen space by a lot.
https://forum.freecad.org/viewtopic.php?t=78431
[image]
[image]
[image]

tcurdt, 2023-05-20:
@MisterMakerNL looking good. But I am more concerned about things like this:

The cursor position:
[image]

Everything red is super hard to read:

[image]

The vertices, the constraints, the dimensions.

luzpaz, 2023-10-11:
Are folks able to track the discussion in this ticket? It's hard to follow. Any ideas on how to make it easier to approach ?

tcurdt, 2023-10-11:
@luzpaz you mean you are not sure how to approach fixing this?

Would you want links to best practices? Or how can we help?

A good start to reading https://venngage.com/blog/accessible-colors/
And there is also the DaltonLens to install as tray application https://github.com/DaltonLens/DaltonLens
Or for a quick check available via web https://daltonlens.org/colorblindness-simulator

I think just saying "yes, we want to fix this" would go a long way. If the FreeCAD team signals willingness to fix this there might also be people helping by working on a new theme.

MisterMakerNL, 2023-10-11:
@luzpaz Yeah, this still on going. I doubt there will be a release party. We just keep going untill the bugs are gone! And there also a ton of devs doing great things too the ui!
I do think the bug tag isn't right, it's more a goal(s) too work too.

luzpaz, 2023-10-11:
Thanks for the responses. I think it would be helpful to hide/resolve issues brought up in this ticket, that way it would be easier to read. 

[image]

tcurdt, 2023-10-11:
@luzpaz I have hard time understanding how hiding information is helping. What comments would you want to hide?
It's not like this is a long ticket.

obelisk79, 2023-10-11:
There is a problem with this ticket. It is poorly defined and doesn't target a specific deficiency. Instead it's gone around in several directions. I think several other recent issues both open and closed much better encompass the spirit of this one and are much more specific with better discussion focus.

By specific deficiency I mean with more specificity than 'default theme is bad'

luzpaz, 2023-10-12:
Agreed @obelisk79
I'm all for the discussion but it's just not conducive for comprehending progress being made.  
 
This ticket should be transferred to a discussion. Please respond with :+1: or :-1: if you agree/disagree

arkadesOrg, 2023-11-04:
okay I finally found how to install themes. And OpenDark looks readable so far. would be great if there was a default mechanism adapting to the system theme. Just to not turn newcomers off too quickly.