FreeCAD/FreeCAD #7629: [Feature Request] [PartDesign] Keep sketch visible during revolution (and possibly other) tasks

pathfinder555, 2022-10-22:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=72779

### Subproject(s) affected?

PartDesign

### Idea description

During the revolution task (additive or subtractive), if you want to select a part of the sketch you are using for the task itself, you currently have to go to the tree view and turn on the sketches visibility, because the sketch will get hidden as soon as the task is started.

I suggest to keep the sketch visible until the task is closed to prevent unnecessary steps (or make the sketch visible when "select reference…" is chosen, but just keeping the sketch visible seems simpler to me).

Given that the revolution preview will be hidden as soon as "select reference…" is selected, there is no risk of accidentally selecting parts of the revolution. I don't se any other potential downsides, apart from a slightly less clean image I guess.

I haven't checked other operations yet, but if the active sketches can be selected for things, I think they should stay visible until the task is completed in these operations as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PLChris, 2022-10-23:
It would even be sufficient to turn visibility on, if "select reference" is chosen.

luzpaz, 2022-10-26:
@pathfinder555 you have a possible solution, can you make a PR and reference this ticket?

pathfinder555, 2022-10-26:
@luzpaz I don't have one, sadly, I can't write code. Someone posted one in the forum thread though:

https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:patch-59

Syres916, 2022-11-04:
Apparently the clarification that @PLChris made isn't good enough for the merger/maintainer so I'm closing the PR.

donovaly, 2022-11-04:
> Apparently the clarification that @PLChris made isn't good enough for the merger/maintainer so I'm closing the PR.

Sorry, but this is no good behavior.

The bug reporter proposed to have the sketch always shown, another poster said it would also be sufficient to show it only when in selection mode. So we need to make a decision. I am more with the bug reporter because it is helpful to see the sketch when creating revolutions.

The second issue is consistency. People complain about this so we should take care. So when we change the behavior, it should be consistently also be there for the other task dialogs of PartDesign, for example also for the Helix feature.

closed 2022-11-05 by wwmayer

donovaly, 2022-11-05:
@wwmayer So you decided only to have this for revolutions? What about the other PD features?

We recently had long debates about consistency and when I start to act accordingly people are pissed. It seems I can do what I want, and it is wrong.

What would have been the correct way here? Should I have merged it, then started a discussion? Or make a PR on my own? or extending the PR?

Can I at least make a PR for Helix?

wwmayer, 2022-11-05:
The forum topic is only about revolutions and if other PD features should be changed then I think people should open new tickets for each of them. I don't like tickets that address several slightly different issues.

donovaly, 2022-11-05:
The point is that users see their particular issue. That is where the inconsistencies come in. Therefore we mergers have the duty to check PRs for this too. And nothing else is what I did. The PR alone was OK.

I will now make a PR to fix the issue for the other PD dialogs.

donovaly, 2022-11-05:
> I will now make a PR to fix the issue for the other PD dialogs.

here it is: https://github.com/FreeCAD/FreeCAD/pull/7722

pathfinder555, 2022-11-05:
> The forum topic is only about revolutions […]

Well, not exactly. I mentioned revolutions specifically because that's where it affects me personally the most, but I kept it open enough to include other tasks. The title of the forum discussion is "keep sketches visible during tasks".

Consistency-wise, there's two ways of thinking here:  either you keep it consistent in the sense that it works in every task where it does serve a purpose (i.e. where parts of the sketch may be selectable) or in the sense that it is visible at all times. There's an argument to be made that the sketch staying visible could be "ugly" if it doesn't serve a purpose, but apart from that, I don't see much of a downside. Actually, since the preview option isn't a thing anymore, It may even be beneficial: on multiple occasions where I switched from the Task window to the treeview, I did forget that I was still in an active task and was confused why FC didn't behave properly. With the sketch still being visible, it may serve as a visual indicator that the task is not done yet.

wwmayer, 2022-11-06:
> There's an argument to be made that the sketch staying visible could be "ugly" if it doesn't serve a purpose, but apart from that, I don't see much of a downside.

For Pad or Pockets there is a certain downside when trying to select a reference axis because choosing any line of a sketch will result into an invalid extrusion.

So, it's fine to discuss for every PD operation where it's beneficial to see the sketch but not to make it visible everywhere because of consistency reasons.

donovaly, 2022-11-06:
@pathfinder555 to conclude: please open a new forum thread to discuss if Sketches that were visible are kept visible when using them for a PartDesign operation. If people agree, open a new issue and list there for what PartDesign features this should be done.

commit 18781bcf (2022-11-05): PD: fixes #7629: Keep sketch visible during revolution tasks
PR #7722 (2022-11-06): [PD] Helix: assure source Sketch is visible when in selection mode
- same as https://github.com/FreeCAD/FreeCAD/commit/18781bcf6 but for Helices (fixes #7629 also for helices)
- also modernize Qt connection
commit 664bc12c (2022-11-06): [PD] Helix: assure source Sketch is visible when in selection mode

- same as https://github.com/FreeCAD/FreeCAD/commit/18781bcf6 but for Helices (fixes #7629 also for helices)
- also modernize Qt connection
commit 9a053b86 (2022-11-06): PD: fixes #7629: Keep sketch visible during revolution tasks