FreeCAD/FreeCAD #7633: [Bug] FreeCAD Contribution policy and process is confusing.

sliptonic, 2022-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=72769

### Version

0.20 (Release)

### Full version info

```shell
N/A
This issue is organizational and affects all versions and the general development process
```

### Issue description

The development process used by the FreeCAD project is not well defined.  It's unclear to new contributors how code contributions are to be structured, submitted, evaluated, and accepted.  This can lead to several undesirable outcomes:
- A contributor spends considerable effort on a feature and then the feature is not accepted.
- Multiple conflicting solutions are implemented resulting in a confusing user experience
- Code contributions languish because no one knows who is responsible for accepting them.
- Potential contributors do not contribute and existing contributors leave the project in frustration.

A process should be established that clearly documents how code contributions will be handled.  The process should be fair and transparent.  The documented process should be put under git version control like any other part of the source so that it can be referenced, changed, and tracked over time.

### Anything else?

Much discussion has been had about the C4 process from the ZeroMQ project.
https://rfc.zeromq.org/spec/42/

When a process is accepted, the wiki including https://wiki.freecadweb.org/Developer_hub  should be reviewed and revised to reflect the process.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-10-23:
Good description of the issue.  Thank you.  
- "When a process is accepted".  Is it too early to start discussing what would constitute acceptance criteria?  
- Could we use the existing C4 process as an "optimistic commit" of the alpha process?

sliptonic, 2022-10-23:
> * Is it too early to start discussing what would constitute acceptance criteria?
I don't think it's too early.  As a starting point, I would suggest acceptance looks like this:  A draft PR is created.  We invite code-review from everyone who's voiced an opinion so far.  They can comment on the text directly.  When all issues are resolved, the PR is marked ready.  Merged by someone other than me.

> * Could we use the existing C4 process as an "optimistic commit" of the alpha process?

That was my original intent.  However, the C4 contract is itself licensed GPL3.  Also, it's written project agnostic so that it can be widely adopted.  I intend to rewrite our CONTRIBUTING.txt document as inspired by C4 but FreeCAD specific.  In places, I'll use language that is more representative of FreeCAD.  Rather than refer to 'the platform' in the abstract, I'll specifically state that we use github, etc.   If we change to another platform later, we just modify our contributing document to reflect that.

berniev, 2022-10-23:
Suggest:
1. Delete "to new contributors ". This applies to everyone.
2. Change "feature" to PR. A PR is not necessarily a feature.
3. For reasons mentioned in the forum discussion I urge not referencing C4's "optimistic merge".

LarryWoestman, 2022-10-23:
Many code contributions are composed of many PR's over time.  A per-PR process may not be what we want so much as a per-contribution process that might be composed of one PR or many.
For example, if there is feedback about the code in one PR, the changes might come in a following PR rather than going back and modifying the first PR.
Many problems are too "big" for a single PR.

sliptonic, 2022-10-23:
> 1. Delete "to new contributors ". This applies to everyone.

Done

> 2\. Change "feature" to PR. A PR is not necessarily a feature.

A PR is a mechanism to get a contribution accepted into a codebase.  The goal of a contributor is to add a feature or fix a bug.

> Many problems are too "big" for a single PR.

One intended side-effect of a process like C4 is to force big problems to be broken down into manageable steps. Can you give an example of a problem that is too big for a single PR and also cannot be decomposed into smaller and more manageable pieces?

berniev, 2022-10-24:
> big issue might need unit testing to be created
FC does not use unit testing. This is perhaps not well understood.

berniev, 2022-10-24:
I find the example a bit confused. Any piece of code is ultimately **testable**, but it may take some refactoring to make it so. That's the risky bit, as this refactoring is done without the protection of unit tests. Then unit tests are created and proved to work. This is not restricted to function level, could be entire class etc. Finally code is changed to address the original (small) piece of the overall 'problem'. 

It sounds like you are seriously considering enforcing unit testing. That would be a big step and a shock to many. May I recommend an excellent book that speaks directly to the issues we are discussing.

> Working Effectively with Legacy Code by Michael C. Feathers ISBN 0-13-117705-2

>

LarryWoestman, 2022-10-24:
A couple of possible examples of "too big for a single PR" are the work I am doing on refactoring the postprocessors and the work that realthunder has done with sensitivity to change that is trying to be merged right now.  I think that the characteristic that makes it hard to decompose those types of things into more manageable pieces is that lots of functionality has to change before the changes start working again.
In the case of refactoring the postprocessors some of the changes cause git to mark pretty much every line of the postprocessor code in multiple files as "changed", which makes any potential reviewer's job hard.
I think it is possible to chop a "big" change into "fragments", but they are often not functional until many "fragments" get comitted, and there is extra work involved in making sure that each fragment is able to build and be tested individually. 
I do think that having a very clear statement of the problem that needs to be solved is important for any change.

LarryWoestman, 2022-10-24:
It is possible to write additional tests one test at a time, with a clear problem statement and a simple PR.
Where it gets "big" is when you move a lot of code to new places, or even just reformatting whole files to match the formatting standards.  Some changes might works as a single PR if they are a focused change even if they end up changing a lot of lines of code.

LarryWoestman, 2022-10-24:
I suspect that the refactoring that rearranged the Path directory structure is another example of a change that was "big".

LarryWoestman, 2022-10-24:
The process of taking some existing code, modifying it to be testable, writing the tests, and verifying that the new version of the code still works properly is hard to break up into smaller pieces unless you do it "off to the side" and don't depend on the modified code until you have all of the work done.

berniev, 2022-10-24:
> examples of "too big for a single PR"

I suggest the problem is not the size of the required change, but the manner of implementation. The RT thing IMHO should never have been allowed to fester and grow like it did. I'm unaware of your big change project, but hopefully we are collectively learning to avoid the 'complicated pr'. One solution for big stuff is to rewrite core to offer an interface or if you like an API and move out a lot of functionality. Big change can now be implented in so-called plug-ins, independent of core. Workbenches sort of have this idea, but increasingly the C++ code base is opened to direct access from python, which undoes the advantage.

>

berniev, 2022-10-24:
> extra work involved in making sure that each fragment is able to build and be tested individually.

Sure! But the (potentially big) reward comes later, when refactoring is much easier.

berniev, 2022-10-24:
> unless you do it "off to the side"

Initial refactoring is by definition risky. Must be done in small, minimal risk, steps, compiling often. But there's no way to avoid it.

berniev, 2022-10-24:
> or even just reformatting whole files to match the formatting standards

Reformatting does not change the functionality of the code. And once we all use it there will be no more pesky formatting warnings or diffs!

berniev, 2022-10-24:
@sliptonic 

> Delete "to new contributors ". This applies to everyone.
Done

Actually, not quite.

Zolko-123, 2022-10-24:
> Rather than refer to 'the platform' in the abstract, I'll specifically state that we use github

Are you sure ? What's the advantage ? I'd suggest to leave "the platform" and specify somewhere that, as of today, it's GitHub. And when that changes, we only have to change 1 line.

berniev, 2022-10-24:
> Are you sure ? What's the advantage ? I'd suggest to leave "the platform" and specify somewhere that, as of today, it's GitHub. And when that changes, we only have to change 1 line.

+1

sliptonic, 2022-10-25:
I read the articles about _optimistic merging_.  I agree that it has benefits, in principle.  However, to make it work, at least three things need to already exist:
1. There needs to be good test coverage and a strong value in the culture for writing/maintaining those tests.  You can't optimistically merge if you don't trust that the test suite will catch regressions.
2. There needs to be a high value for _minimal solutions_ and discipline in the issue->PR process.  If PRs are routinely submitted that are large, unfocused, or not grounded in a known issue, then optimistic merging leads to conflict.
3. There needs to be a shared understanding of the big picture.   You don't need a 100-point roadmap of every feature for the next five years but everyone has to agree on what is important, what's trivial, and the goals of the project.

Based on that, I think _Optimistic Merging_ is something we should work toward but not something we can embrace at this time.  

There's a third way that might be more achievable in the short term.  Let's call it "Reviewer Moderated Optimistic Merging"  The idea is to split _process correctness_ from _code correctness_.  In this model, the maintainers focus on the process but we make that process require sign-off by certain domain experts who comment on _code quality_.  It would work like this:

1. Maintainers are selected based on a history of successful PRs.  They don't need to be expert in any area but have to be comfortable reading code. They are instructed about the process.
4. A PR is submitted.  It's tagged by workbench(s), etc.
5. A maintainer, reviews the PR and invites review by any/all people that he thinks will have input. Those reviewers can invite others.
6. a reasonable time is given for review.
7. The contributor's job is to answer/address any concerns raised by the reviewers.
8. The maintainer's job is to referee, keep the process moving, and click the button when no concerns are left unaddressed. If reviewers make what C4 calls 'value judgments' the other reviewers and the maintainer can step in.

### Example:
Someone submits a PR to make Path Job use links.
OpenBrain takes it as the maintainer.  He invites me because it's a Path thing.  He also invites RealThunder because it involves Links.  Maybe it involves a bit of code I don't understand so I also invite another reviewer.
After a week, I've signed off on it because I'm just awesome that way.
Realthunder shows a better way to do something and the contributor updates the PR.
The third reviewer declines to comment.
OpenBrain merges the PR.

berniev, 2022-10-25:
I'm glad you dropped optimistic merge. A couple of comments.

1. Including "Optimistic Merging" in the name implies a derivative of C4 version, but it is significantly different: C4 is faster than CI, this is days, and is not actually optimistic as it demands a traditional review first. Suggest "Moderated Merge".

2. If a moderator wishes to make 'value judgments' they become an "interested party" and must pass the moderator baton.

berniev, 2022-10-25:
@sliptonic Any change to your view on unit testing? 
> big issue might need unit testing

berniev, 2022-10-25:
A little one ..

> A maintainer, reviews the PR 

Propose: "A maintainer inspects the PR". Review is well defined as something else.

LarryWoestman, 2022-10-25:
I think that we can achieve Reviewer Moderated Merging in the near term, and I like the description by sliptonic above.

It is a much harder task to get people to write more and better tests, whatever kind of tests they end up being :-)

sliptonic, 2022-10-25:
> @sliptonic Any change to your view on unit testing?
> 
> > big issue might need unit testing

No change.   I still think it's a great thing.  I hate the terminology around the whole domain though.   "Functional" testing doesn't test functions but functionality.  "Unit testing" doesn't test units of functionality but tests units of code.   It's like our was designed to be obscure.

sliptonic, 2022-10-25:
> A little one ..
> 
> > A maintainer, reviews the PR
> 
> Propose: "A maintainer inspects the PR". Review is well defined as something else.

Done.  Thanks

berniev, 2022-10-25:
> No change. I still think it's a great thing. I hate the terminology around the whole domain though. "Functional" testing doesn't test functions but functionality. "Unit testing" doesn't test units [...]

No point hating the terminology. Its industry standard.

Your original comment "big issues might need unit testing" infers you see circumstances where unit testing will be required. Can you clarify, as this would be a major.

sliptonic, 2022-10-25:
> No point hating the terminology. Its industry standard.
>
> Your original comment "big issues might need unit testing" infers you see circumstances where unit testing will be required. Can you clarify, [...]

I don't foresee a policy change where we start demanding unit test before you can get a Pr accepted.  But even now I wouldn't accept a new generator in Path that didn't have unit tests. All of the others do.
(To say unit testing doesn't happen in Freecad is wrong. It isn't the norm  but I have specifically started writing unit tests for new code in Path.  IIRC, FEM also is using unit tests.   It's a discipline that is increasingly common among experienced professional software engineers but not well understood by hobbyists.  A lot of contributors to Freecad are in the latter group.)

It's going to take time to turn the boat.  It'll happen faster as we start getting contributions from commercial partners.  For now, the best place to start introducing unit tests is when refactoring problematic code and for entirely new functions and classes.

berniev, 2022-10-25:
@sliptonic Are you referring to the .py files in PathTests folder? I couldn't find any other reference to test files.

sliptonic, 2022-10-25:
> @sliptonic Are you referring to the .py files in PathTests folder? I couldn't find any other reference to test files.

Yes but not all are pure unit tests.  Some are functional (we're still learning) and shine are mixed.  Something like testpathdrillgenerator is more unit test

berniev, 2022-10-25:
Firstly, this is all python stuff where my interest has been C++. So to some extent we may be talking cross-purposes.

Q: Can the test you mention run without all the imports? Will it work if any of the imports are broken?
A YES: You may have a unit test!
A NO: This is not true unit testing. It is testing the unit under test AND a whole bunch of other stuff all at once, which is the antithesis of unit testing.

Proper unit testing completely isolates the unit under test. Dependencies are simulated or mocked (replaced with dummies).

I am in now way deriding the testing you are doing. Any testing is better than none. In the python world your methodology may be industry standard, but don’t think so.

Unit testing is not bleeding edge stuff. The book I recommended earlier was published in 2005.

Randon online: https://machinelearningmastery.com/a-gentle-introduction-to-unit-testing-in-python/ <https://machinelearningmastery.com/a-gentle-introduction-to-unit-testing-in-python/>

Perhaps you can start to appreciate where I’m coming from?

B

PR #7638 (2022-11-02): Policy for FreeCAD Contribution Process (FCP)
fixes #7633

- [ ]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either in `App`, `Base`, `Gui` or one of the `Mod` subfolders. If you need to make changes in several locations, make several pull requests and wait for the first one to be merged before submitting the next ones
- [ ]  In case your pull request does more than just fixing small bugs, make sure you discussed your ideas with other developers on the FreeCAD forum
- [ ]  Your branch is [rebased](https://git-scm.com/docs/ [... cut]
closed 2022-11-02 by WandererFan