FreeCAD/FreeCAD #7639: [Bug] <title>Master Spreadsheet values not updated on linked spreadsheets

PAS-DRAFTER, 2022-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72186

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30694 (Git)
Build type: Release
Branch: master
Hash: 16c52bb18c233a4ed79ac250433327999c8c6a9b
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Curves 0.5.6
  * ExplodedAssembly
  * fasteners 0.4.2
  * Manipulator 1.4.9
  * MOOC 2022.4.21
  * OpticsWorkbench 1.0.8
  * sheetmetal 0.2.56
  * ToolViewReverse_ToolBar
[/code]
```

### Issue description

If I make a change to the Master spreadsheet in my Master file, the changes are not sent to the spreadsheets open in other files.  Additionally, if I do a manual recalculate on the cell in the linked spreadsheet that was suppose to change, that cell updates to the correct value but other cells inside the spreadsheet that use that value do not update unless I manually recalculate each cell individually.

Pressing Refresh does not force the recalculation.  The only way I can get the cells to update is to close the files and then reopen them.

I had this problem occasionally with v19, but it is definitely much worse in v21.  In v19 the problem was usually associated to an error in a formula and once I fixed the formula, it seemed to work again.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-10-25:
Bug pending discussion on forum.

luzpaz, 2023-11-26:
@PAS-DRAFTER can you still reproduce on latest development? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

PAS-DRAFTER, 2023-12-01:
I tested on a little bit using: 

`OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git)
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * fasteners 0.4.53
  * OpticsWorkbench 1.0.9
  * QuickMeasure 2022.10.28
  * sheetmetal 0.2.60`

I opened up one of my old files that has a lot of linking.  If I make a change in my Master spreadsheet, the changes were still not being sent to all the linked spreadsheets.  Even when I went into a spreadsheet on a different file that required an update, the old value was still there.  The Refresh button was grayed out so I couldn't use that. 

If I went to the cell I knew should have changed and right clicked on it and selected Recompute, it would then go and find the new value and the Refresh button would become active.  The problem with this approach is that I would have to know each number that was linked so I could go and recompute each cell that was supposed to have changed.  With a massive set of files, that becomes impossible.  The only way I could usually ensure the changes got set all around was to close the entire project down and reopen it.

To be honest, this had become such of a problem, I have basically gone away from linking multiple files together.  I ran into too many problems where the part on one file would not be right, I'd fix it, and then the next time I loaded the file, it would be wrong again.

I admit that the file structure that I have been using gets quite complex so that might not be a best use case.  I've noticed that this problem doesn't always happen either.  When I have less complex models, the updates seem to work.  I've tried building a "simple" file structure where I could demonstrate this, but I don't think I could ever make the "bug" reliably happen.

closed 2025-01-06 by github-actions[bot]