FreeCAD/FreeCAD #7645: [Bug] Absolute cells in a spreadsheet are not updated

wwmayer, 2022-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72546

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.30735 (Git)
Build type: Debug
Branch: master
Hash: 7c51aa9749eddcfc0981feea3c3468560b88db22
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, Vtk 7.1.1, OCC 7.3.0
Locale: German/Germany (de_DE)
[/code]
```

### Issue description

A reliable procedure to repeat the bug is difficult because if there is only one absolute cell then it is getting updated if there referenced cell changes. But see https://forum.freecadweb.org/viewtopic.php?p=632775#p632775

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-10-26:
See also the German forum: https://forum.freecadweb.org/viewtopic.php?f=13&t=72935

luzpaz, 2024-12-16:
Is this still valid ?

FEA-eng, 2025-08-20:
I can't reproduce it with the file from the German forum but it sounds like something that should be rechecked.

FEA-eng, 2026-01-19:
> [@FEA-eng](https://github.com/FEA-eng), if I enter "$" into a cell it produces an error if not converted into a string (with prepended apostrophe).

Indeed, if I enter this:

`=$`

there's the following error:

`08:30:33  <Spreadsheet> Cell.cpp(675): Unnamed#Spreadsheet.A1: Failed to parse expression.`

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260113 (Git shallow)
Build date: 2026/01/13 22:48:50
Build type: Release
Branch: (HEAD detached at 4cc0a1b)
Hash: 4cc0a1b718735be10261d936066cf1382dd51884
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```