FreeCAD/FreeCAD #7646: [Bug] <title>

kanguru007, 2022-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72868&p=635696

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git) AppImage
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Issue description

1. Create one Body with a simple Box
2. Click on the Box twice, the whole Box is selected
3. Right-click and select "Transform" on the context-menu
3. Drag the arrows and move the Box
4. Later, when saving and reloading the file, or when recomputing the object, the Box jumps back to its previous location

From the discussion in the forum I understood that this option in the context menu is not implemented and should be there (and it does not exist in the Mac version)

I am talking about the option on the menu on the screenshot I attached to the first message in the forum discussion.

### Anything else?

Awesome that there is usable opensource CAD software, and it is looking good!
Keyboard shortcuts for everything would be nice.
Thank you.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-10-28:
In your screenshot, it can be seen that the box is selected, not the Body. What behavior would you expect ?

kanguru007, 2022-10-28:
On the context menu for it the Transform tool is available. I use it and dragging the arrows I can move the box. Later, saving and reloading the file or recomputing the Body, the box reverts to its old position. Regardless of how the software works internally, I would expect that if I'm allowed to select the box and move it then it should stay where I put it.

Roy-043, 2022-11-19:
I can't reproduce the given scenario. If the last feature of the Body is an additive, or subtractive, primitive the effects of Transform are not reverted.

See:
https://forum.freecadweb.org/viewtopic.php?f=8&t=73618

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.30922 (Git)
Build type: Release
Branch: master
Hash: 8ec1279ea8ee32a36fae683b42b5cfc5821734b5
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

kanguru007, 2022-11-19:
As someone said on a later discussion about this, " If you right-click a PD feature in the tree Transform is not available. That seems inconsistent."

Removing Transform from the context menu will fix this issue. This is confusing for people having a first look at FreeCAD and it can be fixed quickly.

adrianinsaval, 2023-03-07:
This likely happens because the box has an attachment, transform only acts on the placement of objects but an attached object will go back to it's attachment position when updated. If you wish to translate an attached object you have to use the attachment offset property or remove the attachment. But it's true that transform probably shouldn't be available for regular PD features.

adrianinsaval, 2023-03-07:
could be considered somewhat related to #6032 and #6238, in general the way we use placement and attachments for PD primitives seems to be problematic.

maxwxyz, 2024-12-21:
This will be solved by the new transform tool

closed 2025-01-13 by chennes