FreeCAD/FreeCAD #7648: [Feature Request] wysiwyg, make going straight into design simply visually easy/logical

goofyseeker311, 2022-10-26:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://github.com/FreeCAD/

### Subproject(s) affected?

Other (specify in description)

### Idea description

more wysiwyg user interface, its not platantly obvious where you sketch, where you extrude, where you view the extrude objects, without looking into a tutorial/help files. check fusion 360 for ui/ux model. you should directly jump into a new sketch-grid display when first starting the problem with a unnamed empty design file. there are no sketch related buttons in the ui-window when creating a new document. must be user friendly wysiwyg not based on learning by tutorials and help files from random ui.

dont require browsing help/forums, if so then ui too complex and barebones, make it more user friendly for everyone to use.

### Anything else?

https://www.autodesk.com/products/fusion-360/features#3d-modeling

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-10-26:
Hi, welcome to the freecad community. Sorry, but I'm gonna have to close this ticket for obvious reasons (i.e too broad and references proprietary examples which is a :no_good:). Please interact on our forum and discuss one feature at a time in separate forum threads. FreeCAD is improving its User Interface although slowly because it's a very passionate subject.

goofyseeker311, 2022-10-26:
so you dont want to make freecad user friendly, even if a clear model example was given, one feature is general user friendliness, are you such a community, I hope so

goofyseeker311, 2022-10-28:
of course not, and currently freecad does nothing for me, for not being user friendly (its just a blank stub of an editor). I'm not actually asking you to make it like fusion 360, I'm suggesting that you make it usable (anything) in the first place. I dont see any cad features in freecad as a non-tutorial/manual wysiwyg potential user. (ie its not a cad program for me, currently).

goofyseeker311, 2022-10-28:
at least fusion 360 is very obvious how its used, in the first place (actions tool bar) which freecad is missing completely, not even searching menus for 30mins.

luzpaz, 2022-10-31:
Sorry it does nothing for you. Right now FreeCAD requires reading documentation and learning its quirks. No way around that, at the moment.

What you are asking is not simple at all (to code and to find community consensus on). It's not a uncommon request, that's for sure )see [pinned tweet](https://twitter.com/FreeCADNews/status/1370482079088148480) on FC's twitter account). In fact we created https://wiki.freecad.org/FreeCAD_Managing_Expectations because of all the aggro that has been directed toward the community all these years. 

* This is not linux neck-beard pride thing. This is complex stuff that needs be proposed, coded, revised, and pass a sort of consensus.  
* We are working on improving the UI/UX  
* These things happen slowly
* If you want to see it happen faster get involved (coding skills are not necessary but there is always something one can contribute)
* Suggest that you sit down and study FreeCAD and get over the learning curve, then you'll have full autonomy over your CAD related work. Use the forum for this.
* If you feel too much aversion, sorry to hear that. Maybe check out solvespace or librecad or one of the cheaper proprietary solutions out there.

howie-j, 2022-10-31:
@goofyseeker311 FreeCAD is made by volunteers, all with different backgrounds, habits and preferences. Of course some prefer simple clean UI's as F360, while other are used to Catia, NX, solid edge etc, and do not mind spending a couple of hours upfront customizing FreeCAD to their liking. There is no one size fits all when it comes to the UI unfortunately.

What i would like with my "Getting started wizard" linked further up, is to give new users a choise between the stock "Full expert mode" experience, and a simpler cleaner UI. If you want to help, you can give my [_Simple Mechanical Configuration_](https://github.com/howie-j/FreeCAD-Wizard/blob/main/configurations/howie_simple_mechanical/howie_simple_mechanical.md) a try, and give feedback on what you would like improved.

I have done a lot of UI cleaning in that configuration, as well as enabling some settings to make it easier to get started (new document and partdesign open by default for example).

Here is a short screencast with how freecad looks in my config. As you can see, very easy to get going. A couple of clicks and you have a 3D model:
[Screencast](https://user-images.githubusercontent.com/82042580/199114667-563f2b24-6a8c-46e5-982f-0c1ab5a07336.webm)