FreeCAD/FreeCAD #7651: [Bug] User cannot generate a helical ramp entry to a profile with G2/G3 gcode.

sliptonic, 2022-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=72730

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.30623 +33 (Git)
Build type: Unknown
Branch: (HEAD detached at upstream/pr/7440)
Hash: c74a0ad3e55cd853668c6ae183a05daa2ec5df16
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.52
  * FeedsAndSpeeds
  * ExtremeProDark 1.0.4
  * FeedsAndSpeeds.bak
  * Dracula 0.0.2
[/code]
```

### Issue description

User Applies a RampDressup to a profile operation.  Then sets the ramp mode to 'Helix'  The generated path is, indeed, a helix but is expressed as an inefficient series of G1 moves rather than G2/G3 arc moves.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sslupsky, 2024-11-07:
I noticed this too.  Generated 23,000 lines of gcode!

luzpaz, 2024-11-07:
@sslupsky can you paste the full [About](https://wiki.freecad.org/About) version info of FC you tested this on ?

sslupsky, 2024-11-07:
@luzpaz Here you go.

OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git)
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * ThreadProfile 1.85.0
  * kicadStepUpMod 10.19.4

Reasonably current version.  I think the issue is simply it generates tons of Gcode because it has to make many small steps to simulate a curve that the G2/G3 codes automate.  So, when you use the the helical ramp entry you get tons of steps.  Some of the stuff I've been doing is corrugated plastic with hundreds of holes.  I stumbled across the helical ramp entry and that makes a nicer hole.  But the amount of Gcode is huge.  The other thing about the helical ramp entry is that is consumes an extraordinary amount of CPU.  I had a prop I was working on today with 189 holes it in.  I had to force quit the app because the operation was taking forever to complete.  Today I was running it on an M2 Max MBP with 96GB RAM.  So, not generally compute limited.  I ran into this issue on my older Mac Pro yesterday as well where I had to terminate the process.  (BTW, it would be nice if there was a way to interrupt FreeCAD and cancel recomputes using the ESC key).

I attempted to use the Helix Drilling operation today but ran into problems because you cannot profile with it.  I posted some comments to a couple other open issues here about that.

luzpaz, 2024-11-08:
@sslupsky thanks for your reports and feedback. Hopefully CAM folks can lean-in to address these things. Although some of them seem quite involved and no small tasks

sslupsky, 2025-06-08:
@luzpaz Was closing this issue due to some automated process that did not get review?  I think this still needs to be addressed at some point.