FreeCAD/FreeCAD #7657: [Bug] line names are wrong between inside Sketcher edit mode and outside Ketcher in 3D view

Zolko-123, 2022-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=22&t=27215

### Version

0.21 (Development)

### Full version info

```shell
Tested with 0.21 master locally compiled and with LinkDaily AppImage:

[code]
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30341 (Git)
Build type: Compiled
Branch: master
Hash: 74ffd403643405ffd7e44e26d9b62673ffa7956a
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.56a
  * CurvedShapes 1.0.3
  * fcgear 1.0.0
  * sheetmetal 0.2.50
  * TabBar
  * Render 2022.2.0
  * Curves 0.5.1
  * fasteners 0.4.4
  * Assembly4_GitHub 0.12.4
[/code]

[code]
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

When using a sketch, the sketch's element's names are displayed both in the 3D window when the element is selected, and in the sketch's task-panel when in edit mode. These names match when a sketch is created, but get out of sync when the sketch is modified.

As an example, create a sketch, draw a radial line to make a clocking reference, and draw a centered hexagon. In this case, the hexagon side in front of the reference line is called `Edge2` both in the 3D view and the sketch edit mode.

[image]
[image]

Now, set the top hexagon line (was `Edge3`) to construction, and the name of the previous hexagon side (the one in front of the reference line) is still `Edge2` in the sketch edit mode, but is now `Edge6` in the 3D view : 

[image]
[image]

You can play around and delete or add new lines, and the result is even more diverging. I made the same test with the LinkDaily branch and the result is the same:

[image]
[image]

This is a very annoying bug because when I have many sketches with many lines, then I look in the 3D window to see which edge/line I have to edit, but when switching the corresponding sketch to edit mode the 3D name doesn't correspond to the 2D entity. 

This is not a topological naming problem, since it persists in the LinkDaily branch. But since the new toponaming merging will affect naming of elements, it is mandatory that it also solves this bug. 

It can be seen in the LinkDaily screenshots that the `g2` reference is identical to the 3D view and the edit view, so it would be logical to make use of that to correctly identify (name) the sketches lines. The logical solution would be to convert the `g2` name to `Line2`, and show that name (`Line2`) in both the 2D edit mode and the 3D view window. 

### Anything else?

This bug has been reported earlier and imported from the previous bugtracker as #5767 ( https://github.com/FreeCAD/FreeCAD/issues/5767 ):

There is also a forum thread about it : https://forum.freecadweb.org/viewtopic.php?f=22&t=27215

I report it here because FreeCAD has been updated quite a lot, and also because I have tested it with the LinkDaily branch, and the bug still persists.

Another forum thread that might be of interest is: https://forum.freecadweb.org/viewtopic.php?f=10&t=70409

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-10-27:
This is not a bug, it's expected behavior.
>  it is mandatory that it also solves this bug.

not at all but I'll agree it would be a nice feature request if the relevant part of the mapped name was shown in the status bar and in the element list instead of the mostly useless indexes

Zolko-123, 2022-10-27:
> expected behavior

This is clearly BS. assigning random names (Edge2, Edge6 ...) to the same entity in different view conditions, or assigning the same name (Edge2) to different entities in different view conditions isn't expected by anyone. If names are given, they **MUST** be homogeneous. 

> so you consciously created a duplicate issue... why?

because [the old one](https://github.com/FreeCAD/FreeCAD/issues/5767) was missing images and was not understandable, and  the [new Contribution process](https://github.com/FreeCAD/FreeCAD/pull/7638/files) clearly asks : 

> the issue SHOULD be complete and accurate and SHOULD NOT require the reader to visit the forum or any other Platform to understand what is being described.

adrianinsaval, 2022-10-27:
> This is clearly BS

It is expected in the sense that it was programmed to be this way, these names where not intended to be matching, but ok, I agree this is problematic behavior and I support the request to show meaningful data (mapped name) in the status bar and the elements list.

> If names are given, they **MUST** be homogeneous.

you keep using _MUST_ for stuff _you want_.
> because [the old one](https://github.com/FreeCAD/FreeCAD/issues/5767) was missing images and was not understandable, and the [new Contribution process](https://github.com/FreeCAD/FreeCAD/pull/7638/files) clearly asks :

you mean the not yet approved process? Everything you say here you could have added as a comment in the existing issue.

Zolko-123, 2022-10-27:
@adrianinsaval : Since you have nothing to say, could you please go trolling somewhere else ?

Zolko-123, 2022-10-28:
I have tested this example with FreeCAD version 0.19 and the behavior is already the same (the naming is toggled with the "_Extended naming_" button):

[image]
[image]

As you can see, the exact same name `Document/Part/Sketch/Edge2` is used for 2 different entities

adrianinsaval, 2022-10-29:
here's how the element list looks like in the toponaming branch:
[image]

@realthunder @abdullahtahiriyo my suggestion for solving this issue is this;
[image]
• forget about what is currently displayed in the 'Name' column, as far as I have seen it doesn't really provide any meaningful information, use that column for the TN algortihm given name (g#, e#) in the mapped column show the geometry name of the corresponding geometry in the sketch's shape, not the useless internal sketcher index.

FlachyJoe, 2022-10-29:
Hi,
As a workaround you can get the sketch geometry index of given sub-objects with this python code:
```
def geometryIndexFromSubObject(sketch, subObjects):
	return [i for (i, geom) in enumerate(sketch.Geometry) for object in subObjects if object.cut(geom.toShape()).Length==0]
```
and vice-versa:
```
def subObjectFromGeometryIndex(sketch, subObjects):
	return [object for geom in sketch.Geometry for object in subObjects if object.cut(geom.toShape()).Length==0]
```

Zolko-123, 2022-10-31:
here is how the element list looked in v0.19 :

[image]

> @realthunder @abdullahtahiriyo my suggestion for solving this issue is this; [image] 

I like this proposal, but then, instead of using `g5` names I propose to replace the "g" part – which stands for geometry I guess – by the element type: Line2, Circle5 ...

[image]

this way, the `Name` column would provide meaningful information. 

(Also, while we're at it, what is the purpose of the _Reference_ column ? It takes up horizontal space for no obvious use)

adrianinsaval, 2022-10-31:
> I like this proposal, but then, instead of using `g5` names I propose to replace the "g" part – which stands for geometry I guess – by the element type: Line2, Circle5 ...

These names are the ones used for toponaming so I don't know if it's desirable to make them that long (side note, external references seem to be named e# instead), also the geometry type is already listed on the first column so there's no real need to repeat that on the second column IMO. The `Reference` column is only meaningful for external references and it shows what geometry it is referencing, IMO this should be shown in a tooltip when hovering above those elements instead as they are rarely relevant and take a lot of space. I'm not too sold on `Flags` column either, does it display anything besides something being construction? don't we already know this by the icon color? In any case this should also be shown in a tooltip IMO.
This widget has to be reworked anyway since it's conflicting with recent changes in master (#7567): 
[image]
IMO only the geometry type, the 3d view name and the mapped name (g2, etc.) are relevant so we could have something like what it is in master with this format: Type (3dview_name , mapped_name), skip the useless list index and relegate the reference and flags data to tooltips when hovering over an element in the list.
Inviting @PaddleStroke to the discussion too since he did recent work on this.

abdullahtahiriyo, 2022-11-06:
That the naming inside edit mode and outside edit mode are different is well know. There is the intention (already from a couple of years), to solve this use the solution RT uses in his fork. This solution is intertwined with toponaming. RT's fork and FC main use a common way of storing a common id to be the same inside and outside edit mode, which is used in RT's fork and is currently unused by FC main, waiting for the code merge. However, this will not come until TP is merged.

Zolko-123, 2022-11-06:
@abdullahtahiriyo : yes, I know, that's why I (also) said that this should be solved with the upcoming toponaming merge. What I wished here is to set a test-case for this toponaming merge. As a summary, there are 2 distinct issues at hand:

1. there are overlapping names in edit and 3D mode, as in [Document.Part.Sketch.Edge2](https://github.com/FreeCAD/FreeCAD/issues/7657#issuecomment-1294888306) pointing to different elements depending on the context. This should ideally be fixed, or alternatively hidden. But showing wrong information to the user is a very bad process. Currently, the same bad behavior exists also in RT's LinkDaily branch
2. the unique ID in RT's branch is something like `g#` (where # is a persistent numerical index), and it seems to work well, but it's not very explicit for the user. I'd like to use that persistent ID with a more user-friendly name like `Line#` or `Arc#` or `Circle#`

abdullahtahiriyo, 2022-11-16:
@Zolko-123 
From what I remember from reviewing toponaming code, this g# is very relevant for toponaming, where it is important to be short. As geometries get the unique ID, and not specific geometry types, I do not think it is possible to implement what you indicate in (2), at least without changing all sketcher toponaming.

adrianinsaval, 2022-11-17:
> 2. the unique ID in RT's branch is something like `g#` (where # is a persistent numerical index), and it seems to work well, but it's not very explicit for the user. I'd like to use that persistent ID with a more user-friendly name like `Line#` or `Arc#` or `Circle#`

but where outside of this dialog would the user ever use this name anyways? AFAIK you cannot reference an edge using this in expressions, this is only used internally to fix the toponaming problem.

Zolko-123, 2022-11-17:
> but where outside of this dialog would the user ever use this name anyways ?

We all have our preferred tools, and mine is the `AttachmentEditor`. Apart from the Sketcher, this is the tool that I use most, for attaching LCS to master sketches and then attach parts to those LCSs. I'll try to explain a simple workflow with screenshots. I'm using the toponaming AppImage from some days ago (FreeCAD-toponaming-20221028.AppImage)

```
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.14555 (Git shallow) AppImage
Build type: Release
Branch: development/toponaming
Hash: d2d0b7f01884bf2c8838c07f4389c9e64fc703b5
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
```

Using only PartDesign and Sketcher, I create a body, a sketch, and an LCS (called _Local_CS_ curiously). I then draw a hexagon in the sketcher, and attach the LCS to geometries of the sketch. This is not an imaginary workflow, this is how _Assembly4_ is used : Model → Sketch → LCS → Link-to-part

[image]

This is where I get to see the reference to _Sketch/Edge6_. If I edit the sketch, this `Edge6` will change and reference other geometries, but thanks to the toponaming thing the `AttachmentEditor` will somehow still manage to keep the correct reference. Now let's look at what happens to `Edge6`, and this is where `g6` comes into play. 

Let's decompress the FCStd file and look at the 2 files _Document.xml_ and _Sketch.Shape.Map.txt_ :

[image]

In _Document.xml_, the attachment support of the LCS (called _Local_CS_) shows that it is referencing in the object `Sketch` the element `Edge6`  (old OCC naming), with its shadow mapping `;g6;SKT.Edge6` (new topo naming):

[image]

This mapping doesn't appear anywhere in _Document.xml_, but appears in _Sketch.Shape.Map.txt_ : 

[image]

But even there, it's impossible to know what it refers-to, this mapping is a black-box. Which means that we are going from the OCC-black-box-naming to a new topo-black-box-naming. And actually it would be easy to have a clear new toponaming : looking in _Document.xml_, the Sketch references its geometries like this : 

[image]

We know that the magic `id="6"` is actually `g6`, and Sketcher knows that this geometry is a line. Since this `id="6"` is not a number but a string, Sketcher could call it `id="Line6"` . And since this `id="6"` is persistent, `id="Line6"` would also be persistent. And then the _AttachmentEditor_ could reference the geometry as `shadow="Line6"` instead of the current garbage. 

And then the _AttachmentEditor_ would have a persistent topo-naming safe reference to attach the LCS to the target Sketch. `Edge6` would still be a moving target kept for compatibility, but `Line6` would be a persistent name. 

AND there would be no need for any mapping because the name would be in the Sketcher itself, making the code simpler and more robust. 

AND the user could have a persistent naming from within the Sketcher editor and outside Sketcher in the 3D window. 

What is not to like ?

> I do not think it is possible to implement what you indicate in (2), at least without changing all sketcher toponaming.

Yes, probably. But I've been assured that the toponaming mechanism (API) and algorithm are independent, so it should not be very difficult. Thank-you for your comments.

adrianinsaval, 2022-11-18:
Ok I agree that would ultimately be nicer to see for the user, but I think it's probably too complex to implement at the moment, and I don't think it would be wise to complicate things by adding geometry type to the topological name, specially considering that the current implementation allows swapping identifiers between different geometry types at the user's command, adding the geometry type to the identifier would break this functionality.
see this example (just remove the zip extension):
[attach_swap_example.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10036661/attach_swap_example.FCStd.zip)
the cube is attached to the arc, if you open the sketch and select a line and the arc then click on
[image]
then exit the sketch you'll the cube follows and is now attached to the selected line.

Also this request goes beyond the scope of the talk we where having, I understand this request was probably your end goal but remember that if you aim for too much you might end up with nothing. What you request is something that simply cannot be limited to sketcher only.
I think the best that could be done _at the moment_ is show in the elements list the corresponding shape edge index, eventually it would be nice to change the displayed identifier in the GUI so it corresponds to a more user friendly name based on the topo name, internally it would likely always need to use the unstable opencascade index. IMO this should be requested in a separate issue and keep this one limited to what can be done in sketcher

Zolko-123, 2022-11-18:
Going a step further in the investigation, if I delete a line in the sketcher, the new attachment support which was _Edge6_ is now _Edge4_ , but `g6` remains persistent : 

[image]

The toponaming stuff does its job, very nice. When looking into the _Sketch.Shape.Map.txt_ file, we get the following : 

[image]

As can be seen, the `g6` entry has changed its order, and is now on place n°4. So what the toponaming in reality does is to provide a look-up-table Edge# <=> g# (and probably also vertices). While in the case of general geometries a simple look-up-table is not enough, for sketcher there is a direct bijections between the input and output information, g# and Edge#. Therefore, this external mapping file could be stored inside the Sketcher properties, as a dict{} may-be, linking any internal sketcher geometry (Line#, Arc#, Circle# ...) to a unique external OCC name Edge#.

> Also this request goes beyond the scope of the talk we where having

You're right, this is the result of the technical discussion we are having here that lead to it

> What you request is something that simply cannot be limited to sketcher only.

Here I disagree: the very point is that Sketcher is behaving and treated differently from other geometrical constructors in FreeCAD, so it could very well be limited to sketcher. 

> I think it's probably too complex to implement at the moment

if it's not done now, before full merge, it will be even more impossible to implement it later. 

> I understand this request was probably your end goal

My end-goal was also a warning : **IF** it's impossible to implement this request into the current toponaming code, **THEN** this toponaming code as-is is bad for FreeCAD because it's an intricated black-box that will be impossible to dis-entangle later. **AND** since this brings some new data structure into the FCStd files, this will have a very long-standing impact : so it's now or never.

abdullahtahiriyo, 2022-11-18:
@Zolko-123 

I have not read the whole thing, but I see you are refer to the persistent ID, which is a geometry unique ID that is stored in the SketchGeometryExtension. You can access in current master this ID like this:

```>>> geo1 = ActiveSketch.Geometry[1]
>>> ext1 = geo1.getExtensionOfType('Sketcher::SketchGeometryExtension')
>>> ext1
<SketchGeometryExtension ("4") >
>>> ext1.Id
4```

I put this ID there, so that RT could use it and to facilitate the merging of toponaming. In master this ID is not persistent on purpose, not to interfere with RT's work and the merge. But if you are using the toponaming branch, there you should be able to get this ID like above if you are using Python, and use it to construct our "Line6" kind of identifiers.

adrianinsaval, 2022-11-19:
> Here I disagree: the very point is that Sketcher is behaving and treated differently from other geometrical constructors in FreeCAD, so it could very well be limited to sketcher.

No it can't, because you are not just talking about what Sketcher does, you're asking for attachment and general 3d view selection to behave differently too, this is a global thing that affects everything that has a shape and placement, this cannot be done in sketcher alone.

> Therefore, this external mapping file could be stored inside the Sketcher properties

I guess it could but I don't see how this is better

> You're right, this is the result of the technical discussion we are having here that lead to it

ok but like i said I think this should be tackled as a separate issue since it goes beyond just sketcher, here we should refocus on sketches UI IMO.

> if it's not done now, before full merge, it will be even more impossible to implement it later.

this is simply false

> IF it's impossible to implement this request into the current toponaming code, THEN this toponaming code as-is is bad for FreeCAD because it's an intricated black-box that will be impossible to dis-entangle later.

false equivalence

Zolko-123, 2022-11-19:
OK, let's agree that we disagree. I wish you good fortune in the wars to come

luzpaz, 2022-11-19:
Thanks for the lively discussion, it shows how important this toponaming merge is. I'd like to keep this open and have some other folk weigh-in. @Zolko-123, above abdullah responded to you (https://github.com/FreeCAD/FreeCAD/issues/7657#issuecomment-1319802511)

maxwxyz, 2024-06-23:
@bgbsww is this still relevant after the recent merges?

maxwxyz, 2024-06-28:
any news on this?

bgbsww, 2024-07-01:
Complicated issue that way predates TNP.  Not really on my radar.  Maybe somebody can investigate the current status?

adrianinsaval, 2024-07-02:
> Complicated issue that way predates TNP

it predates the merge of the algorithm but certainly does not predate the problem and was created around the time we first promised to start merging it. I think it's still relevant. It could IMO be summarized in these comments (obviously biased opinion):
https://github.com/FreeCAD/FreeCAD/issues/7657#issuecomment-1295958916
https://github.com/FreeCAD/FreeCAD/issues/7657#issuecomment-1297570581

it boils down to displaying in some way the topological name and mapped name of sketcher geometry during editing.

On a separate note I think we are missing a pretty neat feature to swap ids between geometry:
https://github.com/FreeCAD/FreeCAD/issues/7657#issuecomment-1319386869

PaddleStroke, 2024-07-03:
I'm not a fan of adding columns as in linkstage, it's overly complex.
What about putting it like this : 
[image]

If the line is not construction/external/internal then we display it's subname.

PaddleStroke, 2024-07-03:
An interesting note about this issue : https://github.com/FreeCAD/FreeCAD/issues/15120#issuecomment-2205553979
The actual sketch.Shape is not updated until edit mode is left.