FreeCAD/FreeCAD #7659: [Bug] Report view is called Output window in settings.

MisterMakerNL, 2022-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=636218#p636218

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30345 (Git)
Build type: Release
Branch: master
Hash: f42a47aa0fd8c4f34ef63c185d9f29b9b6e84737
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0 (Disabled)
  * Dracula 0.0.2
  * ExtremeProDark 2.3.2
  * ExtremeProDark.backup1665346471.659064 2.3.1 (Disabled)
  * ExtremeProLight 1.0.1
  * ExtremeProLight.backup1665559197.7999926
  * FeedsAndSpeeds 0.4.0
  * FEM_FrontISTR 0.1.0
  * IconThemes
  * lattice2 1.0.0
  * PieMenu 1.1.10
  * PieMenu.backup1664822562.2342157
  * ProDarkThemePreferencePack 1.0.0
  * scanObjects
  * SelectorToolbar
  * sheetmetal 0.2.58
  * SP_Startup
  * ToolbarStyle
[/code]
```

### Subproject(s) affected?

Core

### Issue description

Kinda confusing having 2 names. 
I might try to fix this one myself since it is probably just a search and replace fix as far as I understand. 
But what are we going to name it?
In Blender it is called "Info"
But we might want to change the settings name to report view to keep continuity. 
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-11-04:
I suggest going with "Report View" -- in the Forums that's what basically everyone seems to refer to it as. The class itself is even called `ReportView`, so I think it makes good sense to standardize on that. There aren't many occurrences of "output window" in the code, and most of those are comments.

luzpaz, 2022-11-04:
Who wants to tackle this?

PR #7739 (2022-11-08): Gui: Rename 'Output window' to 'Report view'
Closes #7659

- [ ] Requires updating wiki documentation once merged
commit 30b918da (2022-11-08): Gui: Rename 'Output window' to 'Report view'

Closes #7659
closed 2022-11-08 by chennes