FreeCAD/FreeCAD #7666: [Bug] Path WB - Mill Face operation is not set to correct depth

gimpox, 2022-10-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=72988

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 16.04.7 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.30377 (Git) AppImage
Build type: Release
Branch: master
Hash: 2462f4c7a05745488bb9e81ccc7ba0f386f78849
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * fasteners 0.4.16
  * Curves.backup1666112221.8799047
  * sheetmetal.backup1665679793.2838469
  * sheetmetal 0.2.58
  * A2plus 0.4.59e
  * toSketch 1.0.1
  * Curves 0.5.11
[/code]
```

### Issue description

The "Final depth" param (indicating how much deep the operation should go) is not applied correctly.
The maximum depth reached by the operation is always given by

**_step down * k_**

where k is the greatest integer value such that **_(step down * k) < final depth_**.

no matter how much you play with this params.

For example:

(finish step down = 0.0 mm, start depth = 0.0  mm)
final depth = 2 mm
step down = 0.6 mm

--> max depth reached = 0.6 * 3 = 1.8 mm from the upper surface (i.e. 0.2 mm less than the planned depth).

Please refer to the forum for further details.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gimpox, 2022-11-03:
On the forum is marked as "SOLVED", but in this case was not, I simply applied a suggested workaround. The bug is still there.

luzpaz, 2022-11-04:
@gimpox traditionally you also include the workaround in the above reply for posterity

sliptonic, 2022-11-10:
Have you tried this again with the current master?  I took the file from the forum deleted the jobs and recreated as best I could.  It seems to work correctly.
If the problem persists, can you please attach an updated project file showing the bug?

sliptonic, 2022-11-11:
I'm closing this for now.  I'll happily re-open it if shown that the problem still exists.

closed 2022-11-11 by sliptonic

gimpox, 2022-11-11:
New problem here after your patch.
When I open my old file with the build 30922, my object tree is destroyed. If I switch back to my previous build 30377 all is ok.

I attach a picture:
[image]

I attach also my FCstd file:
[Ring_light.zip](https://github.com/FreeCAD/FreeCAD/files/9992147/Ring_light.zip)

sliptonic, 2022-11-11:
This is not a problem but a known side-effect of the source reorganization.  https://forum.freecadweb.org/viewtopic.php?f=15&t=71610

gimpox, 2022-11-11:
OMG... so I have to rebuild the jobs for all my stuff from scratch???
I will stick to version 30377 in that case. I have a life out of this PC...

There is no routine/script that I can launch on Linux to produce a new FCstd file compatible with the new changes?

sliptonic, 2022-11-11:
Yes.  Unfortunately, long term you'll need to rebuild.  I have several large projects in the same condition.  
That's why we tried to give lots of heads up about that change and get it done before a future 1.0 release.

gimpox, 2022-11-11:
Now I'm in troubles... shit... (sorry for the french word).
Please check my reply in
https://forum.freecadweb.org/viewtopic.php?p=640337#p640337

sliptonic, 2022-11-11:
I sympathize.  My plan, when I get around to converting the jobs is to have an old appimage and the current FreeCAD running at the same time and open side by side.   I'll walk through the jobs and operations recreating each one and copying over the settings.  
I will compare the gcode results with the inspect tool until they're the same.
It'll be a pain in the butt but shouldn't take me more than half an hour, even for my largest job.  The property names and types are the same and the GUI largely matches.

One thing that can help is to export the job as a template from the old job.  Then create the new job using the template. This will help bring over tools at least.  More if you modified your setupsheet.

gimpox, 2022-11-11:
Having two Freecad applications on my PC seems the only way to survive. (One for the old projects, and one for the new projects.)
We can continue this discussion on the forum. Maybe you can post there some tricks that could help people to "migrate" in the new "fantasting and astonishing world" of version 1.0 : ) : ) : )