FreeCAD/FreeCAD #7668: [Feature Request] Auto-assign names at datum constraint creation

adrianinsaval, 2022-10-29:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=636963#p636963

### Subproject(s) affected?

Sketcher

### Idea description

This issue is somewhat related to #5965, if but of these issues are solved the constraints usage on expressions would be greatly improved and accessible even to novice users.

Currently sketcher constraints are created without a name unless the user specifies it, this can be problematic when the constraints are referenced in expressions using just the list index as these will change if a constraint with a lower index is removed. My proposed solution to this would be to automatically assign a name to datum constraints when they are created based on constraint type and adding a number, here's my proposed scheme:
[image]
The user can of course write whatever name he prefers in the dialog as always, but this gives an usable name from the get go without effort, if #5965 is also implemented this alone makes carbon copy a lot more reliable.

If for some reason somebody dislikes this behavior we could provide a checkbox in preferences to disable autonaming.

As for the process for generating a name, we take the constraint type and use the corresponding string for it, say `HD` for a horizontal distance constraint then we add a number to it, we check if there are other named constraints with the same format `HD#` and we take the highest number we can find +1.

If somebody feels generous we could also add functionality to auto-name old sketches with constraints that don't have names, but just the suggested change here is enough mitigation for this issue IMO.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-02-13:
@abdullahtahiriyo sorry to bother you but once the grid stuff is done could you look into this topic again? recently I used carbon copy for parametric pipe profiles and this would have saved me some time when I had to remove one of the constraints in the original profile and had forgotten to name another.

abdullahtahiriyo, 2023-02-14:
There is a thread where we discussed a bit the issues (it would make sense to link it here).

I agree that this issue needs to be tackled sooner or later. Yes, deletion of a constraint when using indexing creates problems (even for power users as yourself).

From the discussion, my key learning is: 
"It is possible to make a solution based on naming only. However, you should seriously consider Chrisb's suggestion to create an immutable label per constraint."

An immutable label has the big disadvantage that the expression engine needs to be tweaked. Specially, as I am not knowledgeable in that part of FC. However, it does have the big advantage that it separates naming preferences (given names), from automatic unique id generation.

I have quite some code PR code to review (mostly Paddle PRs) and I expect it will take me some time to go through them. Unfortunately, I will not be able to work on this soon.

adrianinsaval, 2023-02-16:
> There is a thread where we discussed a bit the issues (it would make sense to link it here).

it is linked 😉

> I have quite some code PR code to review (mostly Paddle PRs) and I expect it will take me some time to go through them. Unfortunately, I will not be able to work on this soon.

totally understandable, a million thanks for all your work abdullah! go at your own pace, I'll still be here, maybe a little grumpy from fixing references but still very happy for all I get from freecad 😁

J-Dunn, 2023-04-26:
> I agree that this issue needs to be tackled sooner or later. Yes, deletion of a constraint when using indexing creates problems (even for power users as yourself).

couldn't this whole problem and the horrible topo instability problem be solved by using linked lists ?

adrianinsaval, 2023-04-27:
possibly but that may require too many changes to PropertyConstraintList and related code, auto assigning names could be easier to implement and more flexible for the user.

abdullahtahiriyo, 2023-05-03:
A linked list will not solve the problem that the expression engine works with indices. Naming solves the problem by making the expression engine work with names instead of indices.

adrianinsaval, 2023-05-03:
@abdullahtahiriyo if I where to implement the automatic naming where should I put it? In addConstraint? I doubt it's possible to do in the constraint constructor since it likely has no knowledge of PropertyConstraintList right? Or should it be added to the GUI command only?

abdullahtahiriyo, 2023-05-07:
@adrianinsaval 

The solution is different depending on the exact solution.

If I remember correctly (but feel free to correct me as I talk from memory), the solution having the most support/or better balance, was to suggest constraint names based on sketch-wide uniqueness. In principle, only the UI would provide automatic naming.

I do not know (I would have to check), if it is necessary to always produce a unique index (over the whole history of the sketch), or if it is just enough that gaps are filled.

Horizontal1
Horizontal2
Horizontal3
Delete Horizontal 2
Horizontal4
Delete Horizontal4
==> New suggestion is Horizontal5 or Horizontal4?

This is rather important in the number of resources and accounting you need to make. It is a rather important design decision. If we need Horizontal5 and we set for the other, the code is very different (and it has bigger complexity and less performance). If we can do with Horizontal4, code is simpler. This is an important assumption to validate before starting coding.

I am not familiar with the state in which FC stays when having deleting a named constraint. So which behaviour depends on what we need.

So, two approaches for it, (a) keeping track of indices [Horizontal5], or (b) calculating an unique index per call [Horizontal4]. 

If (a) the only place needs to be sketchobject. The big drawback is that you will need to maintain an accounting that is seldom necessary. Keeping the accounting may be more resource intensive, complex and less performing (think of undo/redo operations too), than just doing the simple match of getting the next index. For this reason, I tend to favour a solution of type (b), if possible technically. In principle, I strongly prefer sketchobject over propertyconstraintlist, because a propertyconstraintlist can be assigned from one sketch to another. Then, the historic maximum value will not be sketch-wide of the sketch being assigned (which is the whole point of this). Also, if we set the scope sketch-wide, the responsibility should belong to sketchobject. Of course, it may be implemented in a helper class ("ConstraintNameAssistant", it may be a nested class) containing all data members necessary for the accounting, sketchobject must notify that class of all changes, including constraint name changes (the user may manually rename a variable matching the automatic naming convention), so that the class can ensure it always stores, for that sketch, the highest historical value per constraint type. It must also handle direct changes to properties from python, so on change, it must record a higher constraint index per type if the user provided one.

So, as it appears, (a) has some complexity (at least intellectually, codewise it may be one function that gets called in all right places), and introduces quite some overhead. In fact, creating a unique name per constraint FreeCAD-wide would probably be easier (just a atomic counter on the constraint class, possibly storing the counter value on save and restoring it on restore). A similar solution is present for geometries, see Part::Geometry (cpp file).

If (b), the place should be under Sketcher/App. Why? Because naming is not UI specific, we may want to have a name suggested for Python users or for FC console only. Which place then? Two main options (i) A new class, something like "ConstraintNameAssistant" (can be nested) or (ii) just a free c style function. I would put this on SketchObject.h (in both cases).

At the end there will be a function (possibly static if in nested class), something like: 
```std::string suggestConstraintName(SketchObject &, ConstraintType)```

Under the assumptions of b) all this function needs to do to produce a sound suggestion is to iterate the Constraint property of the sketch, for each constraint get the current name, match it against a regexp of automatic naming (of the right type), to calculate the maximum index present, and provide the next suggestion. This will NOT ensure that the index is historic maximum of the sketch (it is the assumption of (b)).

From these two subsolutions, (ii) is probably the easiest to implement, given that the Constraint property of a sketch is public. It is reasonably refactorable if you change mind later on.

This, memory not failing, is where I think I was after the long discussion. Of course, you need to validate all assumptions before starting coding, and take into account that I may have not considered everything needs be considered... but that was not your question either... you just asked for a place and got all this long reply instead... sometimes you get unlucky...:)