FreeCAD/FreeCAD #7675: [Bug] Start page called when I do an action in a part

lebearnais64, 2022-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=12&t=73078

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
Installed mods: 
  * PieMenu
  * fcgear 1.0.0
  * TabBar
  * fasteners 0.3.50
  * sheetmetal 0.2.50
  * Curves 0.5.1
  * Assembly4 0.12.2
[/code]
```

### Issue description

When I work on a part, when I do an action, the start page appears and I have to click to the part to see my job.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-01:
What's the workflow to reproduce the bug?  
Has someone on the forum reproduced it ?

lebearnais64, 2022-11-01:
no workflow to reproduce it. it is random.
I notice that when i display the report view i do not have the problem

adrianinsaval, 2022-11-29:
this looks like the same issue as #7745 I suggest closing this one as #7745 is better described, has a bounty and some relevant discussions towards solving this.

luzpaz, 2022-11-29:
Duplicate per :point_up:

closed 2022-11-29 by luzpaz