FreeCAD/FreeCAD #7689: [Bug] Measurement does not support 3 point angle

WandererFan, 2022-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=638250#p638250

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.30767 +7 (Git)
Build type: debug
Branch: robustReferences
Hash: f8572af5db0b71007a8813cf10bf9480409ecbc0
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk , OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * QuickMeasure-main
  * A2plus.backup1662229165.818354
  * A2plus 0.4.56a
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Measurement supports angles between edges, but does not support an angle formed by 3 points.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-11-04:
fixed by pr7694. Closing.

closed 2022-11-04 by WandererFan

luzpaz, 2022-11-04:
Linking to #7694