FreeCAD/FreeCAD #7695: [Bug]No rules on who should merge a PR

WandererFan, 2022-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=638260#p638260

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.30767 +7 (Git)
Build type: debug
Branch: robustReferences
Hash: f8572af5db0b71007a8813cf10bf9480409ecbc0
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk , OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * QuickMeasure-main
  * A2plus.backup1662229165.818354
  * A2plus 0.4.56a
[/code]
```

### Subproject(s) affected?

Project Tools & Websites

### Issue description

There are 8 active "core developers" with merge rights (Maintainers in the terminology of  CONTRIBUTING.md) to the main FreeCAD source repository.

There are currently no guidelines on which Maintainer should merge which PRs.
Historically, the primary Maintainer for a given workbench has handled all PRs related to that workbench.

If the primary Maintainer is otherwise occupied or there is no primary Maintainer for a given workbench, PRs may languish in the queue for extended periods.

a) The contribution policy should be revised to provide guidance on who should merge which PRs.
b) Consideration should be given to increasing the size of the Maintainer pool.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-11-03:
Do you have an idea for how to codify who should merge PRs when the primary developer is absent? We obviously don't want to encode actual developer names into the process document, but we don't have a "sub-developer" or "assistant developer" position, at least not formally.

WandererFan, 2022-11-04:
I don't have a solution in mind.  This was more of a plea to start a discussion.   If the Maintainer is to be a guarantor of the quality of a PR, then I don't see any good solution other than adding headcount.