FreeCAD/FreeCAD #7701: [Bug] Recomputing causes mouse to get locked in rotation-feature

Glitchy-Tozier, 2022-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72947

### Version

0.20 (Release)

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.9.9, Qt 5.15.5, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.2
Locale: English/United Kingdom (en_GB)
```
```

### Issue description

**Steps to Reproduce**

1.    Open attached file
1.    Use Blender-Navigation (it might happen with others, too)
1.    Navigate to Combo-view -> Tree-view (=Model-tab)
1.    Click on KeyNrDependentObjects
1.    In that std_part's Property Editor, inside the Data-tab, scroll all the way down until you find Config -> KeyNumber
1.    Change KeyNumber via the dropdown-menu
1.    Hover over the 3D-view and click on (and immediately release) the middle mouse-button to start rotating the screen
1.    See bug: Once recomputing has finished, you will locked inside the rotation-action

### Anything else?

Not sure what to put into the "**Subproject(s) affected?**"-field. I've left it at "`None`"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-04:
Confirmed on the forum thread

luzpaz, 2022-12-02:
@wwmayer reproduces the issue with a simplified example (in the [thread](https://forum.freecadweb.org/viewtopic.php?p=636598#p636598))

```python
import time
from PySide2 import QtCore

class Feature:
  def __init__(self, fp):
    fp.Proxy = self
  def execute(self, fp):
    time.sleep(1.0)
    print ("Execute")
    QtCore.QCoreApplication.processEvents()

doc = App.newDocument()
obj = doc.addObject("App::FeaturePython", "Feature")
obj.addProperty("App::PropertyFloat", "Number")
Feature(obj)
doc.recompute()
```

> Now select Feature, change its Number property and middle-click inside the 3D view. The problem is caused by the function call `QtCore.QCoreApplication.processEvents()` that causes the system not to send the mouse release event to the navigation style.

Glitchy-Tozier, 2024-12-18:
Still present in:
```
OS: Ubuntu Core 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39651 (Git) Snap 1227
Build type: Release
Branch: main
Hash: e698c73a3a1ec36bf6b735b2cd681d43b95ee0c1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

closed 2025-06-09 by github-actions[bot]