FreeCAD/FreeCAD #7705: [Feature Request] Workbench selector in menu bar.

PaddleStroke, 2022-11-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=72431

### Subproject(s) affected?

Core

### Idea description

Add option to have workbench selector in the menubar.

Advantages:
- Less clutter in toolbar
- wb selector is more visible. For new users it will be easier to find, understand and use. I remember being a bit lost when I created my first freecad file, especially because wb selector is by default after File toolbar, so quite hidden.

### Anything else?

PR opened : https://github.com/FreeCAD/FreeCAD/pull/7679

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-11-13:
> Less clutter in toolbar

but more in the menubar

> wb selector is more visible

please provide proof.

PaddleStroke, 2022-11-13:
This has merged as an option. Please stop wasting time of others when you're not following the topic. This is happening every time you post so it's getting really annoying.

closed 2022-11-13 by PaddleStroke