FreeCAD/FreeCAD #7706: [Feature Request] Document tabs, make background clear.

PaddleStroke, 2022-11-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=73187

### Idea description

Turn the document tab background to transparent. Increasing the 3D view.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NXTler, 2023-07-25:
I would love to see this too!

yorikvanhavre, 2023-07-26:
I personally don't like this too much because I don't like having things in front of the 3D view. I'd make this optional in any case.