FreeCAD/FreeCAD #7709: [Bug] No icon for macros in startpage

adrianinsaval, 2022-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

none, thought this was going to be removed?

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux 5.15.76-1-lts (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30829 (Git)
Build type: None
Branch: master
Hash: 92819be7fd267d3d951061963de8a543c063a57b
Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

unlike FCStd files macros are shown without an icon in the startpage:
[image]

this doesn't look particularly good, it would be nice if an icon for the FCMacro files as is done for the FCStd files that have no preview, perhaps
https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/Icons/MacroEditor.svg
[image]

or
https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/Icons/freecad-doc.svg
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-09:
I think we should make it the 1st one macro editor icon.

luzpaz, 2022-11-09:
ping @yorikvanhavre where is the code for this located ?

yorikvanhavre, 2022-11-17:
Hah i had never seen this before :)
On my system it takes the system icon:
[image]
I think the problem here is that the code that provides a default icon seems weird: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Start/StartPage/StartPage.py#L220

chennes, 2022-12-22:
It would be easy enough to add another condition in the code just above that to specifically handle FCMacro files. We could even consider parsing their metadata (if present), the way we do with the FCStd files.

adrianinsaval, 2022-12-22:
I'm on mobile now and this is just speculation but after thinking about it I suspect that my issue actually comes from using the
`BaseApp/Preferences/Bitmaps/Theme/ThemeSearchPaths` parameter to ignore the system icon theme, the icon theming system needs some rework IMO

PR #8073 (2022-12-22): Start: Add icon for FCMacro files
Detect FCMacro files in Start and display the MacroEditor icon for them. Also uses the Addon Manager's Macro class to parse the Macro metadata and display the author information if available, and makes the image-detection code more robust.

Closes #7709.
commit 9a0e50fb (2022-12-22): Start: Add icon for FCMacro files

Closes #7709
closed 2022-12-22 by yorikvanhavre