FreeCAD/FreeCAD #7746: User/Document/Feature level units selection [Feature Request] <title>

gotchoices, 2022-11-08:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=34&t=34834

### Subproject(s) affected?

Core

### Idea description

This would allow the user to specify preferred display units at three prioritized levels:

- User Preferences (most general default)
- Document (project/part file)
- Feature (individual constraint, dimension, etc)

This shouldn't disrupt the idea of storing feature dimensions internally in a single, standard way (mm) but with this enhancement, the author (not necessarily the current user) would be in charge of how units are displayed on dimensions, features and technical drawings.

This would also solve the matter of this [related feature request](https://github.com/FreeCAD/FreeCAD/issues/7395) and perhaps some others.

Objectives:
- If an author creates a file/model using a certain unit of measure, he can attach that preferred unit to the document (in the properties header) and that will carry along to anyone else who may open the document.
- If an author creates a dimension specifying a particular unit of measure, that unit will stick with the constraint/feature and it will always display with those units regardless of document or user settings.
- A model/drawing could contain multiple dimensions, each displaying with a different unit of measure.  For example, an architectural drawing might use primarily meters, but devolve to cm for smaller features.  A mechanical assembly can contain Imperial/US and metric components and display each differently with its native units.
- The user has a quick, efficient way to specify display units on individual constraints/features.  As an example, one might put a special symbol right after the units when entering (25.375 inch!).  The exclamation point would indicate that the input units should also be the display units.  The user should also be able to modify the display units for existing dimensions.
- Files which specify no unit will fall back on the user preference.  Dimensions that specify no unit will rely on the file default, or then the user preference default.  This should supply backward compatibility.

### Anything else?

I will drop a bounty on bountysource.com in case others also want to contribute to it.  For purposes of claiming the bounty, the standard will be to create the feature request, meet the objectives above, and get the feature integrated into the main codebase.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gotchoices, 2022-11-08:
[Here](https://app.bountysource.com/issues/113499254-user-document-feature-level-units-selection-feature-request-title) is the issue on bountysource.com.

If you're interested in supporting this feature, consider adding to the bounty.

AgCaliva, 2023-01-18:
hi is this available? i want to do this and other bounties. Where can i contact you?

gotchoices, 2023-01-18:
Yes, it is still open.  I haven't used bountysource before so I don't know how to offer much guidance in how you claim it, but I'll cooperate in awarding it if you can implement the feature.  You can also reach out to me on gotchoices.org if you want to collaborate on features, strategy, etc.

AgCaliva, 2023-01-19:
Hi nice then do you have some email or telegram? thank you

luzpaz, 2023-01-19:
Make sure you run the fixes by the devs on the forum instead of independently trying to approach this. There are other '[units](https://github.com/FreeCAD/FreeCAD/labels/Units)' related tickets that need to be considered in the 'big picture'

gotchoices, 2023-01-19:
I prefer not to post email address on a public forum.  Please use the method above.

AgCaliva, 2023-01-19:
I see, then i should do the pull request and you will pay me? i mean i want to know if this is a bounty, if thats the case i will do it. Thank you in advance.

gotchoices, 2023-01-19:
The prize money I pledged has already been put on deposit with bountysource so you claim it by submitting your request through that site, once your work has been completed.  I think I then have to indicate to bountysource whether the requirements have been met or not.  When it is approved, they will release the funds to you.  BTW, others who also like the feature addition can add their own contributions to the bounty so it could get larger.  Might make sense to ask around with others who have been on the forums and see if they will also contribute to this feature.

gotchoices, 2023-01-19:
One other note: I'm happy to communicate off-forum to help you in any way I can.  However, any communication directly applicable to the implementation of this feature should probably happen here in the forum so others can also collaborate and give input and help.

luzpaz, 2023-01-19:
> One other note: I'm happy to communicate off-forum to help you in any way I can. However, any communication directly applicable to the implementation of this feature should probably happen here in the forum so others can also collaborate and give input and help.

This. TIA

gotchoices, 2023-02-25:
Sounds good.  I'm happy to help with testing if you can make builds available for macos.  Make sure you work closely with the community here as Luzpaz has noted so your work is compatible with the work of others on the project.

AgCaliva, 2023-02-25:
Nice then. Yes i wil surely need someone to test thing for macos since i dont think i can get that enviroment. I will write to community in next days to check what has been already done, im reading the forum also, probably will post there too. Thank you in advance.

PaddleStroke, 2023-03-18:
Making the document level unit seems doable and it has been asked several time I seem to recall. Feature level I'm not sure yet, it could be doable or very hard because it might rely on each individual workbench.
Would you split the bounty between both goals?

@AgCaliva are you still working on that? Do you know FC and its source ? I think macros won't do it, you need to do it in the core in cpp.

sliptonic, 2023-03-18:
Not sure if it's useful here but a kind of feature-level unit is possible today.
Example:  create a cube in Part workbench.
In the property pane, enter "2 in" in the length.  The value is converted to 50.8mm.
However, if you click on the expression editor and enter "2 in" as an expression, the property now shows the result and the original expression including the units.  I've used this in the past when I want to retain the original unit input that doesn't match the schema.

[image]

gotchoices, 2023-03-19:
Splitting the bounty seems reasonable.  However, I'm not sure if I can.  Do you know enough about bountysource to know if that can be done after the bounty has been posted?

PaddleStroke, 2023-03-19:
I don't know bountysource. There is nothing to do it on the website? Sounds like a basic feature so it would make sense that it's available.

AgCaliva, 2023-04-04:
Sorry people i have not been able to fulfill this bounty, i think i will leave it to other since i cant afford to use my time in this right now. Really sorry. Next time if im interested in this bounty i will just do pull request. Since then i think im off.
I know im off but as far as i see the related PR mentioned are not completely done the task, it does partially, the PR author needs to read my last post in the forum where i compile everything said. Anyway thats my opinion. I wish you luck.

AgCaliva, 2023-05-09:
Hi people im back and as i said with a PR, but i think i have only solved the part of adding to the Project Files the ProjectUnitSystem setting, which can be also set to be ignored. I will complete the rest of the bounty when im back in a week, but i think i dont fully understand those points:

- If an author creates a dimension specifying a particular unit of measure, that unit will stick with the constraint/feature and it will always display with those units regardless of document or user settings.

Dont fully understand, im not an avid user of FreeCAD so will need your help to find out where a user adds constraints, i mean where i should add this option on the GUI, or where do you suggest me to do it.

So i have completed the first two points mentioned here, and will complete in some weeks(2 or 3) the last:
"Feature (individual constraint, dimension, etc)"

If everything its ok i would like to ask to have some part of the pay since i really need the money.
Thank you in advance!

gotchoices, 2023-05-09:
There is a dialog that comes up when you add or edit a constraint. One 
possibility would be a checkbox there that makes the units "sticky."  
Right now you can enter a dimension using any units and the system will 
accept it, but it will then convert to mm internally and then from then 
on it displays units in the user's preferred units.  Sticky units would 
be remembered per constraint and would always display that way 
regardless of the internal representation or the current user's pref 
setting.

This feature request suggests that units of display are not just the 
current user's preference, but can be specified by the document's author 
as well.

AgCaliva, feel free to contact me off-list by the means mentioned above 
if you want to discuss bounty pay or other non-feature-related issues.

sliptonic, 2023-05-09:
I'm not a big fan of software development bounties.  Like this one, the acceptance criteria is often poorly defined.  It makes it hard for a developer to do the work and claim the bounty.   That said, they do serve a purpose.

@gotchoices,  if you're amenable,  I would suggest considering the acceptance criteria to be these.  If the PR meets the criteria, pay 1/2 of the bounty and pay the remaining 1/2 when the PR is merged.  That gives the developer a strong incentive to work with project maintainers to get the code in a mergeable condition.

1) The user can specify on a per-document basis which unit shema should be used.
2) The user can configure FreeCAD to either respect this setting or to always use the default user-preferred schema
3) When the user opens a document, the application will check if the per-document setting exists and is set and will adjust the application to use that setting.
4) The application will recheck and switch unit schema whenever a document is activated.  So if a user has two documents open with different schemas, when switching back and forth between them, the correct schema is shown.

That is plenty of work for one bounty.  If implemented, it would be a great feature.  
Per-feature units can be done in a work-around way now.  (see my earlier post).  A second bounty for improvement could be offered as well.

gotchoices, 2023-05-11:
@sliptonic, your suggestion has merit.  I could be persuaded, depending on the quality of the work.  As a bounty poster, I will say my primary incentive for posting is to get the feature available to use in my own work.  I also like the idea of supporting the evolution of the project for others.  It is great to code the feature, but if it is done quick and dirty so it is not really suitable for a merge into the core, it isn't much help to me (or anyone else).

AgCaliva, 2023-05-11:
> @sliptonic, your suggestion has merit. I could be persuaded, depending on the quality of the work. As a bounty poster, I will say my primary incentive for posting is to get the feature available to [...]

Im waiting review before asking you to pay me some part, i agree with this, i have never done nothing quick and dirty in my life, if im asked to do so then i prefer to do nothing.
So yeah im waiting feedback in the PR. Next week will start the "per constraint unit system" feature.

sliptonic, 2023-05-12:
> but if it is done quick and dirty so it is not really suitable for a merge into the core, it isn't much help to me (or anyone else).

That's exactly the reason the second payment depends on merge to master.  That way the maintainers have the final say on code quality.

AgCaliva, 2023-05-12:
I mean yeah i agree with that but its hard to get a well defined bounty, it can take too much to start pr, then get feedback and then the dev will know what was wanted exactly. Like in this case i started coding when it was no toolbar with unit system at right bottom, but now there it is and i have to add or migrate same functionality to that one. Thats annoying. In this case its kinda nothing but as i have done a lot of bounties i have found similar situations where that kind of situation happens. A bounty should be something very well defined or the pay will result too low and probably there will be no hunters. In other bounties i have meet mantainers who keeped adding their work to the bounties... it has to be well defined because of that.

AgCaliva, 2023-05-12:
If everything goes ok after this i will do:
https://github.com/FreeCAD/FreeCAD/issues/7745
Since i can solve any bug.

sliptonic, 2023-05-12:
> it has to be well-defined because of that.

Exactly.  I much prefer contract work.  Then the contract is defined in at least three phases:
Phase 1 - A detailed written plan.  the developer does the hard research and figures out what needs to be done.  Gets paid for delivering a plan that anyone else could continue with if the developer dies or quits.
Phase 2 - A working example.  The developer implements the plan and shows it working.  Gets paid for this.
Phase 3 - A merged feature.  The developer works with the community to get the code up to snuff and merged.  Gets paid.

In my mind, this spreads the risk equally between the person offering the contract and the person accepting it.  
It focuses the work down to deliverables.

AgCaliva, 2023-05-17:
@gotchoices hi there sory can i get some part like 100 or 200 us? i will complete the PR asap next days and this week will work on the next part. Talk to me to my mail maybe? ramiel1390@gmail.com Thank you so much.

AgCaliva, 2023-07-03:
I think the PR will be merged so i should start with the other part of the bounty:
"Per constraint Unit System"
So to be sure i understand, where i need to add this feature is when adding/edditing constraints like this?

[image]
[image]

AgCaliva, 2023-07-03:
@gotchoices  BountySources dont pay anymore, and i dont think i will continue with bounty, pay me some part if you can.
https://github.com/bountysource/core/issues/1586
TY.