FreeCAD/FreeCAD #7756: [Feature Request] Tree and Task transparency

PaddleStroke, 2022-11-09:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=34&t=73280

### Subproject(s) affected?

Core

### Idea description

Currently default UI is combobox: It changes between tree view and task view.

RealThunder's branch has a very interesting transparency implementation for the tree view and for the task view.

Advantages : 
- Tree and task boxes are shown at the same time.
- Much more view area, especially for small screens. Even though both tree and task are shown simulatenously.
- Looks better.

You can see how it looks in @MisterMaker screenshot:
[image]

So the idea is to merge the commits from Linkstage branch relating these transparancy settings and make it the default UI.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2022-11-09:
Hmmm. Where is the property pane ? It's of high importance in most workflows.

obelisk79, 2022-11-09:
> Hmmm. Where is the property pane ? It's of high importance in most workflows.

I believe that just happens to be hidden in that screenshot. In LinkStage builds, panels can be configured to "AutoHide" or collapse when not active.

PaddleStroke, 2022-11-09:
> It's of high importance in most workflows.

Most workflow ? I never ever used it. My thoughts are rather that it is not used by most people. But that's another topic because it's not removed by RT. I think there's just a setting to auto-collapse this pane.

PaddleStroke, 2022-11-09:
> > > It's of high importance in most workflows.
> > 
> > 
> > Most workflow ? I never ever used it. My thoughts are rather that it is not used by most people.
> 
> 🙄 And again you don't learn. What's the rationale allowing you to think that your own limited usage of FC (BTW, are doing any real productive work with FC ? I mean, use it, not code in it) may represent standard workflow ? This isn't kind of things you should think about, but listen to. ;)

Sorry what? You're joking right? Please read again you message. Then read again mine.
You say : 

> It's of high importance in most workflows.

By which you're doing exactly what you accuse me of doing. You take your own limited usage of FC, and decide that most workflows use it. You don't even say 'I think that'. No you say : it's like that period.

I questioned your comment, because you just claim to be omnipotent on all the usage of FC, by saying my opinion which is the opposit.

I just said my opinion. I didn't said 'Most people don't use it'. I said I think that. 

So how in the world can you end up posting condescending messages like yours ?

And for your information of course I use parametric 3D, I used it for years to develop commercial products. Why would I be spending time on FC if I wasn't...

PaddleStroke, 2022-11-09:
@obelisk79 could you perhaps make a screenshot using the default UI but with the transparency (I don't have the link branch actually)? Maybe two screenshots, one with the property collapsed and one shown.

oficinerobotica, 2022-11-09:
Please read all and most of all excuse my English. 

Just to be all on the same page I thought I'd give some examples of the overlay while actively using it on a  small project. 
The most important thing to note here is that @realthunder 's implementation doesn't actually remove or modify the current state of the UI but instead just extends the functionality taking advantage of some QT capabilities. 

In fact, on start the interface is indistinguishable from FreeCAD master  the only difference being the highlighted icons in the image bellow. Note that I just didn't want to reset all my personalisation to default for the screenshots but this is just how master would look when using split tree and properties view 

[image]

I think that the first image in  @PaddleStroke  post was not the best to showcase the improvements that the OverlayUI can bring to a workflow. And now that we talk about it, there are so many so called "workflows" and there are also entire books written about man/machine user/interface interaction so I'm sure we will not be able to invent anything new here. The best workflow or user interface disposition is the one that better suits individual user's needs. This is where the OverlayUI shines. In giving yet another option to the user at setting the UI based on it's own needs. And this is IMPORTANT to understand; I (the user) have diferent needs based even on screen size or aspect ratio. 

I , personally,  use a different disposition of the UI on my laptop than on my   ultra wide monitor. Why? Differences in cursor travel distance, font size related to my declining sight, difference in gaze shift etc etc ..... so different needs from even the same user. 

This is why I also think that the OverlayUI implementation from Realthunder would be a emprovement for FreeCAD's UI as it ads another layer of customization that could help users adapt the software to their own personal needs all while not modifying nothing that isn't already a default now in master. It is just an extra UI customization option.

Some examples
1 - enabling the overlay. Please note how the interface seems roomier compared to the screenshot above even tough the panels occupy the same physical space. 

https://user-images.githubusercontent.com/26909793/200936359-234786d7-0817-45cc-9134-ad65d8daedc8.mp4

2- autohide options

https://user-images.githubusercontent.com/26909793/200937624-bb6a152c-5160-499f-be93-a7556867fa7a.mp4

3 - auto show on edit. Note how the panel is automatically opened on a task

https://user-images.githubusercontent.com/26909793/200938043-29edbf63-d1b7-49ef-b06c-e07c97b23b37.mp4

4- drag and drop panels. A feature I use allot on a dual screen setup and I couldn't live wibought now that I tried it
https://user-images.githubusercontent.com/26909793/200938711-464d0f88-a798-4dae-a985-57aa022dc258.mp4

5- If we have to talk about WorKfLow, I often have to deal with assemblies of parts stored in different files so I , personally, prefer a unobstructed view of the tree. The small tree in the combo view is really difficult to read when having a never ending tree. Fusion360 does this even better by moving the feature history of individual bodies in the "timeline" so , in my example it would be even less scrolling. But again this is my personal example and the OverlayUI has permited me to do it in a fast and pretty way.

https://user-images.githubusercontent.com/26909793/200940426-3402b0de-d82f-47e9-aecc-b034d719775f.mp4

The point is that FreeCAD is so vast in usage and capabilities that the point here is not what UI disposition we should use but if we are able to customize it to our needs. For this purpose the OverlayUI would be a highly desirable tool in my opinion.

PaddleStroke, 2022-11-09:
Thanks for the detailed description !
So if we can merge is as it is, it would be a clear benefice, as you stated default doesn't change at all, it only adds customization capacity.
Now the question is, how do we go to merge those? Do you know if it's in RealThunder's plans? Or one of use should cherry pick the right commits? Though I'm afraid that there might be a lot of commits and getting them all + all merge conflicts may be difficult to handle.

oficinerobotica, 2022-11-09:
From my limited understanding and what Realthunder has said in an interview, the overlay code is pretty much self contained and to his telling it should be one of the easier merges (lol)  .... I'm making assumptions and I would like him to weigh in this conversation. 

Having this capability combined with preferences packs in master would be a incredible addition. I strongly belive its worth the hassle.

realthunder, 2022-11-10:
Yes, I plan to submit the changes. Just not sure when. Since now it's brought up, maybe I'll give it a try. The transparency overlay capability alone is kind of limited to just a few files. So it might not be difficult after all. Not sure though.

PaddleStroke, 2022-11-10:
Hi @realthunder,
Sounds awesome. Looking forward testing that PR :)

luzpaz, 2022-11-10:
@realthunder JFYI, @PaddleStroke opened this ticket expressing interest in helping port this feature over. Since there haven't been many folk (AFAICT) that have stepped up to offer this, perhaps there may be room for sharing/discussing this with interested parties (like @paddlestroke) ?

obelisk79, 2022-11-10:
@realthunder @oficinerobotica 
I have used LinkStage exclusively for over a year in the past, so I'm rather familiar with the overlay changes. I found over-time that I came to dislike the Auto-Hide functionality of the panels, would it be possible to add a sort of hide-able drawer feature, where a user could click a widget and have the panel slide out of view and then click to bring back to view as desired instead of relying on mouse hot-zones? In the below screenshot I've mocked up what I think the widget would look like. See the red arrow to understand what I propose for feedback.

[image]

realthunder, 2022-11-13:
@obelisk79 What you are proposing does not seem to be related to the overlay feature, or I am misunderstanding? Did you mean to click the arrow to show the sketch edit panel or the tree view? If it is the tree view, the current implementation has an option to show the tree view only when user click inside a hint area on the edge. It will still auto hide the panel when lose focus. Sure, I can make it stick with a new option, if that's what you mean.

Zolko-123, 2022-11-13:
What is the difference between this and the [Glass addon](https://github.com/triplus/Glass/blob/master/README.md) ?

Zolko-123, 2022-11-13:
> So if we can merge is as it is (...) Now the question is, how do we go to merge those?

No, this is **NOT** the question: the question is "_What does this provide that a long-time existing solution (Glass addon) does not ?_ " Until this is reliable answered I vehemently oppose this merging.

PaddleStroke, 2022-11-13:
> No, this is **NOT** the question: the question is "_What does this provide that a long-time existing solution (Glass addon) does not ?_ " Until this is reliable answered I vehemently oppose this merging.

Please stop wasting other people time when you don't even take the time to read the thread here and on the forum.

Zolko-123, 2022-11-13:
First you'll answer my question: 

> What does this provide that a long-time existing solution (Glass addon) does not ? 

All it says here is that "_Unlike the "Glass" addon, these changes also allow for 'click-through' of the transparent areas of a panel_ " and no, I will not read all the crap that **YOU** write all over the forum. So if that's the only benefit, than I oppose this merge.

PaddleStroke, 2022-11-13:
This is one too much.
If your life is crap you don't have to release your anger on people on the internet. This is just so lame, I would go as far as say that it's pathetic. 
Go outside and get your ass seriously kicked by being such a moron and maybe that habit will pass.
Please stop talking to me in any way.

luzpaz, 2022-11-13:
Hey now.. settle the heck down. Escalation breaks communication down. UI/UX duscussions are always so charged, yikes.

PaddleStroke, 2022-11-13:
For other people's reference, because my initial reply to zolko may look angry and I don't want to look like a jerk : 
'Stop make people waste their time' is a direct quote from zolko. He repeatedly post aggressive content, usually when he has not read anything and insult freely people.
If you want a sample of that you can look here : https://forum.freecadweb.org/viewtopic.php?p=639222#p639222

oficinerobotica, 2022-11-13:
I feel like we are losing focus here .
Let's just try to keep this thread technical so who reads it has an idea of what is usefull to better the experience on freecad. 
Until now there were posts exposing what could be the benefits but of course someone might see disadvantages that we may have overlooked like:
- the default UI is modified(it is not)
- the UI becomes slow (there were some problems at the beginning but Realthunder has quickly solved them and now everything is smooth)
- there are logic inconsistencies in the UI behaviour using overlay (personaly I haven't found them)
- the code is garbage ( I'm not the one to judge but I highly doubt it) 

We are shooting our self's in the foot. There is usable code that might bring improvements and it would be nice to know the technical reason why it shouldn't other than the lack of programming man power. 

I'm really no one and my opinion counts close to nothing in the Freecad programming world but at my day job only the results count and opportunities are taken instead of being wasted all for the common goal.

MisterMakerNL, 2022-11-13:
@realthunder already did all the hard work! Mucho gracias!
So I'd say just implement it so everyone can test it in the dev branch. 
That's what the dev branch(0.21) is for isn't it?
Maybe it has bugs maybe not, you don't make any progress if you are afraid of making mistakes!
Although as a realthunder fanboy, he probably has this under control.💗

luzpaz, 2022-11-14:
>  So I'd say just implement it so everyone can test it in the dev branch.

FWIW, I agree.

PaddleStroke, 2022-11-14:
Me too.
Implementing as an option (without modifying default ui) have no valid counter argument.

The only debate that could be held is : should it be the default UI. My answer to that is : Let's first merge as an option. And when it's confirmed to work flawlessly, we can run a poll on the default ui.

0penBrain, 2022-11-14:
As long as it is an option (disabled by default) this is probably good to merge.
One thing that has to be checked before is how new code is intricated with current one (basically weigh how hard that will be to roll back if it has to be done in the future). ;)

Zolko-123, 2022-11-14:
This violates 2 fundamental FreeCAD and open-source principles:

- as per the [new contribution policy](https://github.com/FreeCAD/FreeCAD/pull/7638/files) line 65:

> 3. The User or Contributor SHOULD write the issue by describing the problem they face or observe. Links to the forum or other resources are permitted but the issue SHOULD be complete and accurate and SHOULD NOT require the reader to visit the forum or any other Platform to understand what is being described.

There is no problem involved, no description of anything, AND the contributor wants to link to the forum which is explicitly forbidden (SHOULD NOT).

- it drives towards code-duplication, where an existing solution is ignored for a new implimentation

And I'll add a third argument:

- the existing Glass implementation is better because the text has a clear transparent background, where-as this new implementation has text on solid background and is ugly.

@oficinerobotica :
> We are shooting our self's in the foot. There is usable code that might bring improvements 

exactly: there is already an existing implementation, it has been around for years, you personally have produced nice variants of it, I want to know why it doesn't solve the problem.

So, first, **describe the problem you want to solve**. The argument that some code exists somewhere is not enough to merge it.

MisterMakerNL, 2022-11-14:
@Zolko-123 realthunder version has stylesheet support.

realthunder, 2022-11-14:
> * it drives towards code-duplication, where an existing solution is ignored for a new implimentation
> 
I wouldn't say ignored. My code draws inspiration from the glass Addon. And I acknownledged that in my forum post about my implementation.

> And I'll add a third argument:
> 
> * the existing Glass implementation is better because the text has a clear transparent background, where-as this new implementation has text on solid background and is ugly.

This should actually be my counter argument. Because the Glass implementation has no option to have any kind of background, the text is often difficult to read, while mine has option for solid background, translucent background, text outline/shadow, or no background if you prefer. You see, it is actually not that easy to have a proper background.

> * it drives towards code-duplication, where an existing solution is ignored for a new implimentation

You seem to have a tendency of drawing conclusion without knowing the ins and outs. Have you read the code of the Glass implementation? Have you read the code of mine? What makes you think there is 'code duplication'? You don't even know what are the options I provide.

There are a few other extra features comparing to Glass. There is transparent mouse pass through, offered as an option, like everything else. The overlay supports all dockable panels, not just the tree view, on all four sides. User can freely drag panels into overlay dock region, non-overlay dock region, or float

PaddleStroke, 2022-11-14:
Another advantage is that it's not an addon. I just tried to install glass addon, and on restart was greeted by error messages, a different stylesheet than I used and it was not working. I guess it needs some setup. Build in features just works.

Please go ahead @realthunder, everyone but one is supporting this.

realthunder, 2022-11-14:
I might need to split the changes into two PRs, the first of which would be some supporting code changes in the Base module.

luzpaz, 2022-11-14:
AFAICT glass addon hasn't really been improved or hacked on besides realthunder's work. He stepped up and did the work. All of it is optional to the end user to enable or disregard as they prefer. I'm +1

obelisk79, 2022-11-15:
> @obelisk79 It will still auto hide the panel when lose focus. Sure, I can make it stick with a new option, if that's what you mean.

I found the display and hiding with mouse focus and 'hot zones' on the edges of the screen to be a bit troublesome when manually editing properties and moving between the display and the panel. So, yes the idea is for a widget on the edge of the panel to allow me to easily click to both hide and display and have it stick with or without focus was my idea. I was hoping it would be something simple to implement since the core display/hide behavior is already implemented in your code.

luzpaz, 2023-09-17:
Keep an eye on #10681

PR #7888 (2023-10-11): Gui: overlay dock widgets user interface
Closes #7756 

Forum thread https://forum.freecadweb.org/viewtopic.php?f=34&t=73280
closed 2023-10-11 by yorikvanhavre