FreeCAD/FreeCAD #7757: [Problem] Pad/Extrude geometry failure

tonyhain, 2022-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.21
  * freecad.gears 1.0.0
  * kicadStepUpMod 10.16.5
  * lattice2 1.0.0
  * Rocket 2.8.4
  * SlopedPlanesMacro
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
[/code]
```

### Problem description

[Feather_Boards.zip](https://github.com/FreeCAD/FreeCAD/files/9973945/Feather_Boards.zip)
[image]
both PartDesign/Pad & Part/Extrude fail to produce evenly spaced 4mm fingers with 1mm gaps on 120 degree angle. Executing a cut between both approaches results in everything disappearing, so it would appear they use the same code to generate the fill.

### Anything else?

I am still learning FreeCad, and having various problems with pockets and gaps in geometric fills, but this one shows very clearly that the geometry engine is getting confused on non-right angles. 

I would guess there are rounding errors somewhere because one of the hexagon patterns locked the process for 18 hours (until killed) because the sketch claimed "fully constrained", but I had not started by ensuring complete wire closure before defining the angles. When there were 1800 constraints it was impossible to find the points that had not been defined coincident, and even if known, finding the angle constraint that needed to be deleted would have been very tedious. I realize that new feature requests should be submitted separately, but if there was a simple way to get all the constraints associated with a selected point or line, the diagnostic process would be much easier. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2022-11-09:
You have added some overlapping geometry on the 3rd teeth. So it clearly a user error, please someone close this ticket.

2 things:

 - This is the weirdes way to do a geometry with a pattern. Use PartDesign Pattern, with that you spare around 97% of the work you hve done here.

 - This is not helpdesk, for questions please use the FreeCAD form.

Thanks

tonyhain, 2022-11-10:
Thanks for the reply.
1) I was not expecting help desk support here
2) the noted overlap explains the Cura output but not the convoluted display in FreeCad. If the shading is intended to identify overlap, that color pattern is not intuitive as a geometry overlap.
3) I have had nothing but failure using the Part Design Pattern tool (starting with it's unidirectional function), so I switched to Sketcher copy because I thought it would be faster.
(Clone fails to copy Spreadsheet Constraints in other desigs while Copy at least converts them to constants) 
4) I regularly see lines do 180 degree inversion during Concatenate, so that is likely the source of this one. This one the line was short enough that I didn't see it happen, and the constraint noise obscured the result. Actually, the 120 degree constraint got copied as 1200 as well.

Removing the overlaps did produce the expected result, so close the ticket.. I do apologize for the errant report.

> You have added some overlapping geometry on the 3rd teeth. So it clearly a user error, please someone close this ticket.
> 
> 2 things:
* > This is the weirdes way to do a geometry with a pattern. Use PartDesign Pattern, with that you spare around 97% of the work you hve done here.
* > This is not helpdesk, for questions please use the FreeCAD form.

closed 2022-11-10 by luzpaz