FreeCAD/FreeCAD #7758: [Problem]  Regression caused by the Path rearranging of a few weeks ago.

LarryWoestman, 2022-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30921 +1 (Git)
Build type: Release
Branch: staging
Hash: 2a0e865bbd4b252c1aed4713623ec86c110beeca
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Path

### Problem description

The Path rearrangement of a few weeks ago broke "./FreeCAD -t PathTests.TestPathPost".

I think that I am the only person actively using that test since it is used to compare the original and "refactored" versions of some postprocessors to make sure that they produce the same results.  This test is not run in the normal Path tests, but is important for my work in refactoring the postprocessors.

I am also submitting a PR with the fixes for this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-16:
@LarryWoestman and @mlampert where are we on this and #7759 ?

LarryWoestman, 2022-11-17:
PR #7759 has been pulled by sliptonic.  I have built and tested the master branch today and all of the tests passed, including PathTests.TestPathPost.
This issue can be closed.

closed 2022-11-17 by LarryWoestman