FreeCAD/FreeCAD #7764: [Issue] Rendering bug when changing parameter

mmatthe, 2022-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.2 LTS (awesome)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Issue description

In attached file [Spiegel.zip](https://github.com/FreeCAD/FreeCAD/files/9977997/Spiegel.zip), when I undock the document view and change the parameter "NutOffset" from 40mm to 20mm, the rendering becomes erroneous. If I change the amount only by 10mm, the rendering remains fine, hence I can change from 40mm to 30mm to 20mm and it's fine, but I cannot change it from 40mm to 20mm immediately. The same holds for the opposite direction (from 20mm to 40mm). See accompanying video for an explanation. 

Also, there are errors in the console window and the report window which I cannot understand and that might be the root cause of the issue. 

```
Sketcher::SketchObject: Link(s) to object(s) 'Pocket' go out of the allowed scope 'Sketch005'. Instead, the linked object(s) reside within 'Part'.
Sketcher::SketchObject: Link(s) to object(s) 'Pocket001' go out of the allowed scope 'Sketch007'. Instead, the linked object(s) reside within 'Part'.
Sketcher::SketchObject: Link(s) to object(s) 'Pocket' go out of the allowed scope 'Sketch005'. Instead, the linked object(s) reside within 'Part'.
Sketcher::SketchObject: Link(s) to object(s) 'Pocket001' go out of the allowed scope 'Sketch007'. Instead, the linked object(s) reside within 'Part'.
68.7679 <App> Document.cpp(3553): Spiegel#Sketch005 still touched after recompute
68.7679 <App> Document.cpp(3553): Spiegel#Pocket still touched after recompute
68.7679 <App> Document.cpp(3553): Spiegel#Sketch002 still touched after recompute
```

[Peek 2022-11-10 07-26.webm](https://user-images.githubusercontent.com/16668802/201018379-511a42c1-88d2-4b23-a13e-0440eb6f377a.webm)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2022-11-10:
You have two circular references, espacally on the changed parameters. See dependency graph. So it is an user error.

Besides this is not a helpdesk, for user questions please use the forum. Thanks.

luzpaz, 2022-11-10:
@mmatthe to understand the issue better open a discussion and ask on the Help forum. If you indeed did find a bug please re-open this ticket.

closed 2022-11-10 by luzpaz