FreeCAD/FreeCAD #7765: Slab has an invalid Shape [Issue]

hu-stefan, 2022-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Issue description

I want to create an slab with the arch workbench.

I use the structure tool an select the following parameter:
[image]

After inserting the Slab on the Report view an Waring shows up.
Slab has an invalid shape.
The Slab doesn´t get represented but is part of the model.
[image]

I tried to create an slab via the sketcher workbench and after creating the structure this slab get shown although it has the same Structure Parameter in Height, Length and Width.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-10:
I can reproduce this on but is this the intended workflow ? CC @paullee0 

OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30921 (Git) Snap 417
Build type: Release
Branch: master
Hash: 7acc67ea43797441948b405cd0e925775b62eadf
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * TabBar
  * ExtremeProDark 2.3.3
  * FreeCAD-BlenderBIM 2022.10.20
```

paullee0, 2022-11-11:
Indeed, not much description about this workflow in the wiki
https://wiki.freecadweb.org/Arch_Structure

Probably broken somehow at some point :D

luzpaz, 2022-11-13:
We probably need a unit test for this function as well in order to avoid regressions in the future

yorikvanhavre, 2022-11-17:
Confirmed. The other precast tools seem to work, only the slab is defect.
The UI is also not intuitive, one needs to go in column mode to place a slab...

Roy-043, 2022-11-24:
The tool is not intended for the creation of cuboid prefab slabs, but for hollow core slabs. Reasonable values must be entered in the "Precast elements" section of the task panel. Accepting the default values (all zero) leads to the mentioned problem.
[image]

So the bug is combination of faulty default values, faulty user input, and the failure of the tool to respond gracefully.

I don't think there is any regression here.

Roy-043, 2022-11-27:
The simplest way to solve this would be to use a default value if "Slab base" is zero. Using 50mm or Height/4 (whichever is smallest) can work.

BTW I agree that the UI is confusing. You can even create a staircase with the tool!

Roy-043, 2022-11-27:
I have removed the Regression label.

paullee0, 2022-11-28:
Interesting, it calls ArchPreset rather than ArchStructure to create those objects

Roy-043, 2023-01-01:
As already mentioned, the code in ArchPrecast.py can create several objects. For all objects the handling of faulty values is currently:
1. When the object is created it does not receive a shape.
2. If the object already exists its current shape is kept. Which is confusing IMO.

Changing this behavior for Prefab Slabs only would not be consistent.

For a slab the current checks are:
https://github.com/FreeCAD/FreeCAD/blob/780d8416e24d5d2ada1c9a9521171515489c820f/src/Mod/Arch/ArchPrecast.py#L560-L570

My proposal is to change this to (I have removed the unnecessary brackets):
```
        import Part

        if length == 0 or width == 0 or height == 0 or base == 0:
            obj.Shape = Part.Shape() # Null shape.
            return
        if width <= 2 * slant:
            obj.Shape = Part.Shape()
            return
        if height <= base:
            obj.Shape = Part.Shape()
            return
        if holenumber > 0 and (holespacing == 0 or holemajor == 0 or holeminor == 0):
            obj.Shape = Part.Shape()
            return
        if holemajor < holeminor:
            obj.Shape = Part.Shape()
            return
```
In this proposal a check for the value of SlabBase (base) is included. And in case of faulty values the result is always a null shape. `obj.Shape = Part.Shape()` would then also have to be implemented for the checks of other precast objects.

PR #8287 (2023-01-31): Arch: Fix ArchPrecast.py issues
This PR fixes #7765.

Additionally:
* Introduced a 'null shape' warning in case of invalid property values. The behavior is similar to f.e. Arch_Wall. If the object does not yet exist no shape is created. If the object already exists the old shape is retained. In both case the warning is displayed.
* Input fields in the 2nd section of the task panel now show units. This avoids the problem where, if a comma was used as the decimal separator, values would by multiplied by 10 after each use of the command.

- [x]  Your Pull Request meets the requirements outlined i [... cut]
closed 2023-01-31 by yorikvanhavre