FreeCAD/FreeCAD #7772: [bug] display -> colors -> simplecolor does not work (v0.20.1 on macos 13.0)

TomMuc1, 2022-11-10:


luzpaz, 2022-11-10:
Transferred this ticket from FreeCAD-Bundle to main FC repo because this may not be related to Conda build only.

luzpaz, 2022-11-16:
@chennes if you have macos, can you test this issue?

ipatch, 2022-11-19:
just tested this on macos mojave. i am unable to reproduce.

```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.30032 (Git)
Build type: Release
Branch: ipatch.builds.0.20.1.release
Hash: bbdc3f94105a132aab1c7df270a8e0c3b03f78fb
Python 3.10.6, Qt 5.15.5, Coin 4.0.0, Vtk 8.2.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Plot 2022.4.17
  * CfdOF 1.17.7
  * Defeaturing
  * A2plus 0.4.56a
  * Manipulator 1.4.9
  * fasteners 0.4.3
  * Help 1.0.3
  * Curves 0.5.6
```

---

[image]

luzpaz, 2022-11-20:
Thanks @ipatch !

luzpaz, 2022-11-29:
Can someone reproduce on the weekly Conda macos build  ?

luzpaz, 2022-12-01:
@PLChris  @chennes ?

PLChris, 2022-12-09:
I used to have this issue since the FreeCAD 0.16 times on a Mac using High Sierra (whatever number that was).
I can no longer reproduce here with macOS 13.0 (Ventura).

luzpaz, 2023-02-19:
@chennes when you get a moment, can you try to reproduce on macOS ?

chennes, 2023-02-19:
I'm self-compiled, and haven't been able to reproduce this issue. Today that's an ARM build:
```
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.0.31927 (Git)
Build type: Release
Branch: cMakelibfmt
Hash: 29660e5a61f0ad586866cad675ed9665666ccca4
Python 3.10.9, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

chennes, 2023-02-19:
Also works fine with Conda-latest:
```
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: C/Default (C)
```

chennes, 2023-02-19:
I've closed this since no one has been able to reproduce -- if it crop back up again please re-open. Thanks!

closed 2023-02-19 by chennes