FreeCAD/FreeCAD #7787: [Issue] FreeCAD 0.20.1 crashes on startup on MacOS 11.7.1 Big Sur

WelchITConsulting, 2022-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20.1
```

### Issue description

When attempting to start FreeCAD the application crashes and generates a fault report which gives no clue as to why the application has crashed. I therefore ran the executable from the command line to get the FreeCAD debug messages which are included below:

```
# packages in environment at /Users/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/osx/APP/FreeCAD.app/Contents/Resources:
#
# Name                    Version                   Build  Channel
aom                       3.4.0                hb486fe8_1    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
arpack                    3.7.0                hefb7bc6_2    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
blas                      2.116                  openblas    conda-forge
blas-devel                3.9.0           16_osx64_openblas    conda-forge
blinker                   1.4                        py_1    conda-forge
boost                     1.74.0          py310h509978a_5    conda-forge
boost-cpp                 1.74.0               h8b082ac_8    conda-forge
brotli                    1.0.9                h5eb16cf_7    conda-forge
brotli-bin                1.0.9                h5eb16cf_7    conda-forge
brotlipy                  0.7.0           py310h1961e1f_1004    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2022.6.15            h033912b_0    conda-forge
cairo                     1.16.0            h56d7d38_1012    conda-forge
calculix                  2.20                 h61d0cde_0    conda-forge
certifi                   2022.6.15       py310h2ec42d9_0    conda-forge
cffi                      1.15.1          py310h96bbf6e_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
coin3d                    4.0.0             hf16527e_1011    conda-forge
cryptography              37.0.4          py310h52c3658_0    conda-forge
curl                      7.83.1               h372c54d_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310h2ec42d9_0    conda-forge
double-conversion         3.2.0                he49afe7_0    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
ezdxf                     0.18            py310habb735a_0    conda-forge
ffmpeg                    4.4.2           gpl_h5a1d76f_107    conda-forge
flann                     1.9.1             h30321d8_1010    conda-forge
fltk                      1.3.8                h7ffa276_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               h676cef8_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.36.0          py310h3c08dca_0    conda-forge
freecad                   0.20.1          py310h41becee_1    conda-forge
freeimage                 3.18.0              haafd79f_10    conda-forge
freetype                  2.12.1               h3f81eb7_0    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.8               hb161b9c_0    conda-forge
gettext                   0.19.8.1          hd1a6beb_1008    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
gitdb                     4.0.9              pyhd8ed1ab_0    conda-forge
[... 190 lines cut ...]
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.6                h775f41a_0    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
zlib                      1.2.12               hfe4f2af_2    conda-forge
zstd                      1.5.2                hfa58983_4    conda-forge
FreeCAD 0.20.1, Libs: 0.20.1R29410 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[1111/171934.023367:ERROR:icu_util.cc(251)] Couldn't mmap icu data file
./FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 22893 Trace/BPT trap: 5       "${PREFIX}/bin/freecad" $@

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-13:
@WelchITConsulting how are you installing FreeCAD, it looks like you're using the conda build, right? Please run FreeCAD from the command line `/path/to/FreeCAD --version --verbose` and paste.

Looks like it may be associated with Qt specifically qt5webengine ? 
Links:  
https://bugzilla.redhat.com/show_bug.cgi?id=1920379  
https://github.com/Homebrew/homebrew-core/issues/104008

luzpaz, 2022-11-13:
> Did you deplay the application using macdeployqt? If so, then please check if you have DYLD_FRAMEWORK_PATH set in your environment variables and unset it. Deploying the application and specifying a DYLD_FRAMEWORK_PATH to the Qt libraries seem to conflict.

From [Qt bugtracker](https://bugreports.qt.io/browse/QTBUG-85576?focusedCommentId=519859&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel)  

CC @looooo

WelchITConsulting, 2022-11-13:
I installed FreeCAD using the FreeCAD_0.20-1-2022-08-20-conda-macOS-x86_64-py310.dmg package available on the website download page.

The information I previously provided was from executing the application from the command line as trying to execute it graphically just failed without error. I have run it again using the version and verbose switches as specified which gave a popup with the following content

```
FreeCAD 0.20.1 Revision: 29410 (Git)

Library versions:
boost    1_74
Coin3D   4.0.0
Eigen3   3.4.0
OCC      7.6.2
Qt       5.15.4
Python   3.10.5
PySide   5.15.5
shiboken 5.15.5
VTK      9.1.0
xerces-c 3.2.3
```
The command line displayed the version list as provided in my first message, without the last error message.
I have checked the environment and the DYLD_FRAMEWORK_PATH is not set, unless it gets set within the FreeCaD executable.

luzpaz, 2022-11-15:
@ipatch have you seen this before (i know this is conda related but I wonder if you've seen it on homebrew ?)

luzpaz, 2022-11-15:
@WelchITConsulting see https://github.com/FreeCAD/FreeCAD/issues/7482#issuecomment-1280097970

luzpaz, 2022-11-15:
This is a dupe of #7482

closed 2022-11-15 by luzpaz