FreeCAD/FreeCAD #7790: [Problem] User can't easily create gcode to countersink/counterbore holes

sliptonic, 2022-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.30921 +1 (Git)
Build type: Unknown
Branch: bug/7600
Hash: 91a25ef3a288b2930ed572032c3167099849d110
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.52
  * FeedsAndSpeeds 0.4.0
  * ExtremeProDark 1.0.4
  * FeedsAndSpeeds.bak
  * Dracula 0.0.2
[/code]
```

### Subproject(s) affected?

Path

### Problem description

The Path workbench has a drilling operation but does not have a countersink or counterbore operation. 
Countersinking and counterboring are similar to drilling but workflow may vary in important ways.

The current workaround is to use the drilling operation.  This requires the user to manually set the start and end depths.  This is not intuitive.

### Anything else?

Forum discussion was started here:  https://forum.freecadweb.org/viewtopic.php?f=8&t=73354

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2022-11-12:
First let's talk about the workflow.
Can we expect that the user will model the countersink/counterbore?

If it isn't modeled directly, what parameters would the user need to configure in a task panel?

GeneGH, 2022-11-12:
I think the goal of the Path WB should be to match the FC model. Various dressups, such as dogbones and tabs are either functions that attempt to better match the model or that attempt to facilitate manufacturing.

Countersinks and counterbores are neither. They should be modeled directly.

With that said, there may still be extra processing performed by the Path WB to help match the drawn features.

luvtofish, 2022-12-10:
I’m currently struggling this now and can see valid reasons to model / not model. The former provides a quick visual reference check while giving the model a polished finished look. The later OTOH, could speed up the modeling process if it were to also update the model with the user defined countersink/counterbore specs In a dialog much like the thread tap in the drilling op. Thus achieving the polished, finished look of the former. 

Countersinks have a couple popular included angles depending on which screw spec is being used. The dialog should, much like typical mechanical drawings, request the depth and major diameter of the countersink including the angle. 

For CBore there are standards established for the size of SHCS and their corresponding hole dimensions. The dialog for it could simply ask for the screw size and model the specs for depth and diameter. If the cbore is not  a bolt head hole then it should allow for custom user defined dimensions for depth / diameter.

The answer to this could be as simple as “either can work”. If it’s modeled and a dialog is called at a later stage, delete what was user modeled and implement what was entered in the dialog with a visual update to the model. If I had to choose one over the other I’d say go with the not user modeled approach but update model after dialog entry.

 cheers

Dan

tcurdt, 2024-05-18:
I can see both sides of the argument. But while it is a little odd to not model countersinks on more complex models it could be a tremendous time safer making some simpler (2D style) jobs even simpler.

Just my 2 cents.

closed 2025-07-05 by sliptonic