FreeCAD/FreeCAD #7791: [Issue] Linear Pattern With Single Occurrence Fails

n3ws, 2022-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/Ireland (en_IE)
[/code]
```

### Issue description

Setting the number of occurrences to 1 causes Recompute failed error.
According to the documentation, LinearPattern with 1 occurrence is allowed:
https://wiki.freecadweb.org/PartDesign_LinearPattern 

For me this is important because it breaks my parametric design where the
number of supporting structures is calculated from the length of the part.
For small parts, a single support would be fine.

Attached minimal example.

### Anything else?

[linear_pattern_single_occurrence.zip](https://github.com/FreeCAD/FreeCAD/files/9997086/linear_pattern_single_occurrence.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2022-11-13:
@luzpaz 

Even i the documentation it is not noticed that the occurrence 1 is allowed, never was, since you have divide through it, which makes the result invalid. So this is no bug and works as expected. This is a feature request.

Thanks.

n3ws, 2022-11-13:
Upon rereading the documentation I see that it talks about features, 
not occurrences. Sorry about mixing those. Please consider adding
this as special case so that single occurrence simply does nothing,
because it makes sense for parametric designs.

n3ws, 2022-11-13:
Here is a relevant discussion from Freecad forum:

[https://forum.freecadweb.org/viewtopic.php?style=10&f=8&t=54085&start=10](url)

It appears that single occurrence was working in version 0.19.23578
Maybe this is a regression?

grubertechnic, 2022-11-17:
> It appears that single occurrence was working in version 0.19.23578
> Maybe this is a regression?

The question is why it worked before. Maybe there was a error in the function and the catch was no function at all. So it still works as is should do, so no bug at all. I do not say this is an invalid feature request.

But also i the thread you linked, the solution is would be to deactivate a PartDesign feature. In my opinion, i would also prefer this way. It is just not implemented.

n3ws, 2022-11-17:
I got around the issue by using MultiTransform. No problem about
single occurrence anymore.

closed 2022-12-27 by wwmayer

mfraser, 2023-01-18:
I'm getting this errorL
QObject::connect: No such signal Gui::UIntSpinBox::valueChanged(uint)
QObject::connect: (sender name: 'spinOccurrences')
https://forum.freecadweb.org/viewtopic.php?f=3&t=75360, is this related?

donovaly, 2023-01-19:
> I'm getting this errorL QObject::connect: No such signal Gui::UIntSpinBox::valueChanged(uint) QObject::connect: (sender name: 'spinOccurrences') https://forum.freecadweb.org/viewtopic.php?f=3&t=75360, is this related?

Already fixed: https://forum.freecadweb.org/viewtopic.php?p=654188#p654188

luzpaz, 2023-01-19:
The actual commit by wwmayer: 12ed13ddbea

commit c26a4c79 (2022-12-27): PD: fixes #7791: Linear Pattern With Single Occurrence Fails
commit 76aca425 (2022-12-29): PD: fixes #7791: Linear Pattern With Single Occurrence Fails