FreeCAD/FreeCAD #7793: constexpr names should not be UPPER_CASE

berniev, 2022-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Issue description

Macros and constexpr are very different animals. 

UPPER_CASE constexpr names are misleading, and a range of other issues.

e.g. "Do not use ALL_CAPS for constants just because constants used to be macros." : [coreGuidlines](https://github.com/isocpp/CppCoreGuidelines/blob/master/CppCoreGuidelines.md#es9-avoid-all_caps-names)

LibraryVersions.h (generated file) seems the only current upper case constexpr definitions so is of limited impact right now, but should IMHO be fixed before becoming a precedent.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-11-19:
@berniev your PR has been merged. Can this ticket be closed or do want to create further PRs?

berniev, 2022-11-19:
@wwmayer Thank you! Yes it can be closed.

closed 2022-11-19 by berniev