FreeCAD/FreeCAD #7795: [Issue] Path: Tool paths don't stay where they should

FesixGermany, 2022-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Issue description

When using Path workbench and choosing a work coordinate system that is not aligned (offset and rotation) with FreeCAD's coordinate system and origin, the tool paths do not stay where they are created. They are all aligned to this origin wich in most cases makes it unable to see which path belongs to what feature.

Picture below shows tool paths aligned with features while something is edited within the corresponding Path job.
[image]

Picture below shows tool paths not aligning with features when outside the corresponding Path job.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-14:
Thank you, can you upload an example file (you can zip it and drag drop it here) TIA

FesixGermany, 2022-11-14:
I think I got where the problem is after some more testing.
I am very new to the Path workbench, as far as I understand now a job creates a copy of the selected solid which then is aligned to the global origin wich then is used as the work coordinate system. This is why all tool paths start at the global origin when they are displayed.
This is very confusing for me, I would like to have a new coordinate system placed relative to the object to be machined, rather then a copy created which then is moved to the global origin.
I have a quite simple piece to be machined and displaying multiple tool paths already is extremely chaotic and confusing.

luzpaz, 2022-11-14:
@FesixGermany please open a forum thread to discuss this issue. x-post back here.

luzpaz, 2022-11-14:
This may be [Topological naming problem](https://wiki.freecad.org/Topological_naming_problem). Please confirm your issue on the forum before opening a ticket. Closing ticket for now. If this is a valid bug, please re-open.

closed 2022-11-14 by luzpaz