FreeCAD/FreeCAD #7805: [Problem]

ANDREMIV, 2022-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: makepkg
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.12.4
  * Curves 0.5.13
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
  * 3DfindIT 1.2.0
[/code]
```

### Problem description

Bug in sketcher copy function in sketcher :
Bug reproduction:

1)Open fresh freecad
2)Part design -> new part -> new sketch - XY
3)In sketcher, create arc of circle, fix angle of arc of circle with angle constraint in sketcher
4) Select arc of circle with rectangular box
5) Select copy sketecher function in sketecher, try to make copy
6) Copy is not working properly, buggy parrallel constraints are added

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-16:
Have you discussed with anyone on the forum to see if they can recreate ?

adrianinsaval, 2022-11-23:
this was fixed by #7822
OP did not post in the forum that he had created a github issue, @ANDREMIV please always crosslink when you post in both places

closed 2022-11-29 by luzpaz