FreeCAD/FreeCAD #7810: A2plus: Cannot Suppress the Constraint in the data field. Suppressed field is grayed out

DanMiel, 2022-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30858 (Git)
Build type: Release
Branch: master
Hash: 7a5019b2625a148acb23fb915df48ffa69636fe9
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60b
  * Assembly4 0.12.4
  * BIM 2021.12.0
  * Curves 0.5.11
  * fasteners 0.4.16
  * Manipulator 1.4.9
  * Moveparts 2022.10.28
  * QuickMeasure 2022.10.28
[/code]
```

### Problem description

After constraints are made between part in A2plus the constraint can normally be suppressed in the data field in the Combo View. The field is a True/False but is grayed out.
Dan 

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-16:
Is this a A2plus issue though?

DanMiel, 2022-11-16:
Yes, it is an A2plus issue.

luzpaz, 2022-11-16:
Closing this ticket and opened https://github.com/kbwbe/A2plus/issues/543 in the A2plus repo.

closed 2022-11-16 by luzpaz