FreeCAD/FreeCAD #7811: [Issue] Crash when editing denominator of fraction of inch.

ebalduf, 2022-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.4.23
  * Assembly4 0.12.4
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

When using "Building US (ft-in/sqft/cft)" and you edit dimensions with ft and a fraction of an inch (no actual inches), specifically the denominator FreeCAD crashes.  Specifically edit 1' 1/4" to 1' 1/2" .. Crash!  Very Reproducible. See Anything else section for stack trace. 

### Anything else?

{"app_name":"freecad","timestamp":"2022-11-15 21:37:06.00 -0700","app_version":"","slice_uuid":"19790ace-2ee8-370f-9def-5edcac95c345","build_version":"","platform":1,"share_with_app_devs":0,"is_first_party":1,"bug_type":"309","os_version":"macOS 12.6.1 (21G217)","incident_id":"9153687E-3FF5-4402-8129-35B26B0B9150","name":"freecad"}
{
  "uptime" : 73000,
  "procLaunch" : "2022-11-15 21:36:05.0201 -0700",
  "procRole" : "Foreground",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "MacBookPro16,2",
  "procStartAbsTime" : 73377425754126,
  "coalitionID" : 2452,
  "osVersion" : {
    "train" : "macOS 12.6.1",
    "build" : "21G217",
    "releaseType" : "User"
  },
  "captureTime" : "2022-11-15 21:37:05.9572 -0700",
  "incident" : "9153687E-3FF5-4402-8129-35B26B0B9150",
  "bug_type" : "309",
  "pid" : 27265,
  "procExitAbsTime" : 73438361878664,
  "cpuType" : "X86-64",
  "procName" : "freecad",
  "procPath" : "\/Applications\/FreeCAD-20.1.app\/Contents\/Resources\/bin\/freecad",
  "parentProc" : "Exited process",
  "parentPid" : 27259,
  "crashReporterKey" : "29521505-AA21-4CFA-9A91-EFD59D46187A",
  "responsiblePid" : 27259,
  "wakeTime" : 3167,
  "bridgeVersion" : {"build":"20P411","train":"7.0"},
  "sleepWakeUUID" : "8FEFF24A-5BCF-4383-8BC8-E59535832974",
  "sip" : "enabled",
  "isCorpse" : 1,
  "exception" : {"codes":"0x0000000000000000, 0x0000000000000000","rawCodes":[0,0],"type":"EXC_CRASH","signal":"SIGABRT"},
[... 159 lines cut ...]
USB Device: USB 10/100/1000 LAN
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Backup+  Desk M
USB Device: Anker USB-C Basic Dock
USB Device: USB 2.0 Hub
USB Device: hub_device
USB Device: USB Keyboard
USB Device: USB Optical Mouse
USB Device: Anker USB-C Basic Dock
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 86.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 86.0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-16:
Hey @0penBrain I can't test ATM. Do you know if this a known issue?

0penBrain, 2022-11-17:
Not that I know. Will have a look.
Would need details though.
Where to edit? In Sketcher directly or in Property pane?
Which character to delete when editing? Only the 4 or also the double quote first?

0penBrain, 2022-11-17:
Definitely can't find how to reproduce. Really need step by step procedure.

grubertechnic, 2022-11-17:
Can also not confirm it (latest master on Debian11).

ebalduf, 2022-11-17:
This is on a mac, not sure if that matters.  Below is a macro record of my sequence up to the point just before the crash.  Once you run the below, then double click on the dimension (make sure you are in US units), and backspace over the denominator in the dimension (ie the 2 in[ 1' 1/2"] ) crash. 

# -*- coding: utf-8 -*-

# Macro Begin: /Users/ebalduf/Library/Preferences/FreeCAD/Macro/reproduce_bug.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign
import PartDesignGui
import Sketcher

# Gui.runCommand('Std_DlgMacroRecord',0)
### Begin command Std_New
App.newDocument("Unnamed")
# App.setActiveDocument("Unnamed")
# App.ActiveDocument=App.getDocument("Unnamed")
# Gui.ActiveDocument=Gui.getDocument("Unnamed")
# Gui.activeDocument().activeView().viewDefaultOrientation()
### End command Std_New
# Gui.runCommand('Std_OrthographicCamera',1)
### Begin command Std_Part
App.activeDocument().Tip = App.activeDocument().addObject('App::Part','Part')
App.activeDocument().Part.Label = 'Part'
# Gui.activateView('Gui::View3DInventor', True)
# Gui.activeView().setActiveObject('part', App.activeDocument().Part)
App.ActiveDocument.recompute()
### End command Std_Part
### Begin command PartDesign_Body
App.activeDocument().addObject('PartDesign::Body','Body')
# import PartDesignGui
# Gui.activateView('Gui::View3DInventor', True)
# Gui.activeView().setActiveObject('pdbody', App.activeDocument().Body)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection(App.ActiveDocument.Body)
App.activeDocument().Part.addObject(App.ActiveDocument.Body)
App.ActiveDocument.recompute()
### End command PartDesign_Body
# Gui.Selection.addSelection('Unnamed','Body')
# Gui.runCommand('PartDesign_NewSketch',0)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Part','Body.Origin001.XY_Plane001.')
App.getDocument('Unnamed').getObject('Body').newObject('Sketcher::SketchObject','Sketch')
App.getDocument('Unnamed').getObject('Sketch').Support = (App.getDocument('Unnamed').getObject('XY_Plane001'),[''])
App.getDocument('Unnamed').getObject('Sketch').MapMode = 'FlatFace'
App.ActiveDocument.recompute()
# Gui.getDocument('Unnamed').setEdit(App.getDocument('Unnamed').getObject('Body'), 0, 'Sketch.')
# import Show
# ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
# tv = Show.TempoVis(App.ActiveDocument, tag= ActiveSketch.ViewObject.TypeId)
# ActiveSketch.ViewObject.TempoVis = tv
# if ActiveSketch.ViewObject.EditingWorkbench:
#   tv.activateWorkbench(ActiveSketch.ViewObject.EditingWorkbench)
# if ActiveSketch.ViewObject.HideDependent:
#   tv.hide(tv.get_all_dependent(App.getDocument('Unnamed').getObject('Body'), 'Sketch.'))
# if ActiveSketch.ViewObject.ShowSupport:
#   tv.show([ref[0] for ref in ActiveSketch.Support if not ref[0].isDerivedFrom("PartDesign::Plane")])
# if ActiveSketch.ViewObject.ShowLinks:
#   tv.show([ref[0] for ref in ActiveSketch.ExternalGeometry])
# tv.sketchClipPlane(ActiveSketch, ActiveSketch.ViewObject.SectionView)
# tv.hide(ActiveSketch)
# del(tv)
# del(ActiveSketch)
#
# ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
# if ActiveSketch.ViewObject.RestoreCamera:
#   ActiveSketch.ViewObject.TempoVis.saveCamera()
#   if ActiveSketch.ViewObject.ForceOrtho:
#     ActiveSketch.ViewObject.Document.ActiveView.setCameraType('Orthographic')
#
# Gui.Selection.clearSelection()
# Gui.runCommand('Sketcher_CreateLine',0)
App.getDocument('Unnamed').getObject('Sketch').addGeometry(Part.LineSegment(App.Vector(-52.067940,17.946821,0),App.Vector(-17.946825,32.570164,0)),False)
# Gui.runCommand('Sketcher_ConstrainDistanceX',0)
# Gui.Selection.addSelection('Unnamed','Part','Body.Sketch.RootPoint')
# Gui.Selection.addSelection('Unnamed','Part','Body.Sketch.Vertex1',-52.0679,17.9468,0)
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('DistanceX',0,1,-1,1,52.067940))
App.getDocument('Unnamed').getObject('Sketch').setDatum(0,App.Units.Quantity('317.500000 mm'))
# Gui.Selection.clearSelection()
# Macro End: /Users/ebalduf/Library/Preferences/FreeCAD/Macro/reproduce_bug.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++

0penBrain, 2022-11-17:
OK, I reproduced with 0.20 AppImage (I have no crash but an unhandled exception, which means same).
Bu we have another issue... This unit schema is totally broken in current dev... :confused:

webmite, 2022-12-30:
I believe this is related to #7124 .

adrianinsaval, 2023-01-02:
if anything that could have fixed the issue... AFAIK that change is only present in 0.21.30752 and above, you reported this problem in 0.20.1

AjinkyaDahale, 2023-05-09:
I'm not able to add things like `1' 2"`. The feet are understood, but inches are not.
```
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33014 +5 (Git)
Build type: Debug
Branch: sketcher-general-tangency-splines (no edits in units)
Hash: c74c87f52e49b7045678444a3c7439e16831054a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/India (en_IN)
Installed mods: 
  * Assembly4 0.50.1
  * DynamicData 2.46.0
  * Curves 0.6.8
```

JohnOCFII, 2023-08-13:
I was not able to reproduce this crash.  I DO have to enter the measurements with a plus(+) sign, otherwise they are not accepted.  Not sure if that behavior changed (and may have been related to the reported crash).

```1' + 1/2"``` as opposed to entering ```1' 1/2"``` as suggested in the initial note.

[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

luzpaz, 2023-08-16:
@ebalduf can you reproduce crash ?

adrianinsaval, 2023-08-17:
feel free to reopen if you can reproduce but since others have reported not reproducing this I'm closing it. The way the parsing of this is done changed a lot in the 0.21 dev cycle so I highly doubt this is still present