FreeCAD/FreeCAD #7812: [Problem] Menu "Formteil" is missing in english version

grautvornix, 2022-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Problem description

If language is set to German, you have the menu "Formteil".
After switching language to english, this menu is missing

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-16:
Example of #6704  
CC @0penBrain @donovaly

0penBrain, 2022-11-16:
@luzpaz Not really an example of the other issue (the other is about translations not being updated directly at language change).
Moreover I don't see issue reported here. The German 'Formteil' is strictly same as English 'Part' menu... :confused:

luzpaz, 2022-11-16:
> (the other is about translations not being updated directly at language change)   

I see what you're saying. But @0penBrain that wasn't my impression of the whole issue. I understood it as anomalies would present when setting in a new language. This falls under that behavior as well, IMHO

0penBrain, 2022-11-17:
@grautvornix Can you please give some screenshots/screencast ?

luzpaz, 2022-11-17:
> Can you please give some screenshots/screencast ? 

@grautvornix indeed, i could use some help recreating this.

luzpaz, 2022-12-20:
@grautvornix ping

luzpaz, 2022-12-29:
@Kuzma30 can you recreate this ?

luzpaz, 2023-02-06:
Closing as no response from op

closed 2023-02-06 by luzpaz