FreeCAD/FreeCAD #7818: [Problem] FreeCAD produce different results under the same input parameters

SYehChen, 2022-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/Taiwan (zh_TW)
Installed mods: 
  * CfdOF 1.16.1
[/code]
```

### Problem description

I have a very simple testing model here. It shows that FreeCAD FEM will produce different results under the same input parameters

The original model is 01.FCStd. The CAD parameters are linked to the Spreadsheet. After changing the input parameters in the Spreadsheet several times, and changed them back the the original value of  02.FCStd, FreeCAD produce difference results from its FEM result (for example, von Mises stress).

I suspected this behavior long time ago, so I decided to make a simple model to reproduce the problem

Unless I have done something wrong here, this is a serious problem of FreeCAD.

[model01.zip](https://github.com/FreeCAD/FreeCAD/files/10029019/model01.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2022-11-17:
I can not confirm it, the result is still absolute the same when i edit the Spreadsheet and rerun the calculation. But there are some questions in it.

Do you recalculate the mesh after the editing? I did not for testing purposes. But when you do it, gmsh can have randoms in the mesh in it. When you rerun gmsh, you will see very little differences. I am sure they are intent (start at random point), also i am sure there is also a difference when you use different amount of CPUs.

Also note, the real difference of the value of the result is the minimal value. Since the shape have discontinuous (rectangle corners), a very little difference of the shape seems to have a impact, but the value are still low and neglectable. In the maximal value the discontinuous shape have also a little bit of impact, but is in the tolerance of a FEM calculation.

grubertechnic, 2022-11-17:
> In meshing programs there are certainly radom seeds, especially for freee meshing. But the random number is usually psuedo, not really random.

Then it is not random. Random means not reproduceable at all for the usecase.

> So if Gmsh will produce different results every time even when the inputs are the same, then maybe we should notify the developer of Gmsh.

They are, look at the documentaion, see https://gmsh.info/doc/texinfo/gmsh.html#index-Mesh_002eRandomSeed. Gmsh is a beast. and offers many function.

> In some industries, there are even regulations enforcing the consistency of results from FEA program.

True.

> The bottom line is, an engineer program should produce the same result when then inputs are the same.

Also the will be always a tolerance in computation, especially in FEM.

SYehChen, 2022-12-07:
> Then it is not random. Random means not reproduceable at all for the usecase.
>
>
> They are, look at the documentaion, see https://gmsh.info/doc/texinfo/gmsh.html#index-Mesh_002eRandomSeed. Gmsh is a [...]

I have been using commercial FEA for about 30 years, and never seen this situation before. Netgen doesn't seem to have this problem either (or the discrepency is much much smaller)

luzpaz, 2022-12-21:
@grubertechnic when you report your findings trying to reproduce bugs, please always post your full [About](https://wiki.freecadweb.org/About) info or else your feedback is severely lacking important data.

adrianinsaval, 2023-05-11:
how much difference are we talking about here?

SYehChen, 2023-06-09:
Usually 1%~2%. Sometimes it can be 5%.

FEA-eng, 2024-02-18:
@luzpaz or @maxwxyz I don't think there's anything that could be fixed in FreeCAD itself here so I would close this issue.

closed 2024-02-19 by maxwxyz