FreeCAD/FreeCAD #7830: [Problem] [TechDraw]3d Dimensions are too difficult to create

WandererFan, 2022-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Techdraw

### Problem description

a) To create a dimension based on 3d geometry requires the use of the awkward LinkDimension command.   It should as close as possible to the method of creating a dimension based on projected 2d geometry.

b)  The code to create dimensions is duplicated for each type of dimension.  The common bits should be extracted.

c) Geometry references (PropertyLinkSubList) are awkward to use.  A layer should be added to minimize the complexity of retrieving the geometry details.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-03:
@WandererFan is this ticket still relevant ?

WandererFan, 2023-03-03:
closed as completed by #8141

closed 2023-03-03 by luzpaz