FreeCAD/FreeCAD #7841: [Issue]

SarenT, 2022-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 1397818c9664d158281339a84be59d3e9887f6ad
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, Vtk 7.1.1, OCC 7.5.2
Locale: English/UnitedStates (en_US)
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Issue description

Setting an empty cell alias does not work anymore via the alias text field.
Steps to reproduce:

1. In an empty project, create a new spreadsheet
2. Select a random cell
3. Type a valid alias name into the alias field
[image]
5. Hit `enter` key on the keyboard.

Result: Nothing happens.

Expected Result: Cell turns to yellow and alias is set.

This used to work in 0.19v.

Workaround:

1. In an empty project, create a new spreadsheet
2. Select a random cell
3. Type in a value into content field e.g. =1mm
4. Hit enter to set the value
5. Select the same cell again
7. Type a valid alias name into the alias field
[image]
8. Hit `enter` key on the keyboard.
9. Select another cell to see the alias was set (cell turns to yellow).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-11-23:
This works from the cell properties dialog but not from the main window

closed 2022-11-28 by chennes

donovaly, 2022-11-28:
@chennes , something to be backported right? If so, can you please to the backport?

chennes, 2022-11-28:
Sure, I've done it.

PR #7887 (2022-11-28): [Spreadsheet] Fix alias setting for empty cells
Fix #7841

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the features added or c [... cut]
commit cba90dd9 (2022-11-28): [Spreadsheet] Fix alias setting for empty cells

Fix #7841
commit d17c9e35 (2022-11-28): [Spreadsheet] Fix alias setting for empty cells

Fix #7841