FreeCAD/FreeCAD #7844: Set stylesheet not loaded at startup (Ubuntu 22.10 / Snap)

EelcoA, 2022-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=325625#p325625

### Version

0.21 (Development)

### Full version info

```shell
The changed stylesheet is not loaded at startup. It is saved in the ~/snap/freecad-mechanical/common/user.cfg. The workaround mentioned in the forum discussion doesn't work. I tried changing the user.cfg by adding the path to the stylesheet, but that didn't work either. Other settings (like language) are loaded.
```

### Issue description

1. Change stylesheet (Edit, Preferences, Stylesheet) in another stylesheet
2. Restart Freecad
3. Set stylesheet not loaded

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-20:
Full [About](https://wiki.freecad.org/About) info needed

luzpaz, 2022-11-20:
Can you test an offiicial FreeCAD snap? (you're using https://snapcraft.io/freecad-mechanical which is a 3rd party packager)

luzpaz, 2022-11-20:
Can't recreate on latest official snap 

```shell
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.31058 (Git) Snap 439
Build type: Release
Branch: master
Hash: e6c53d8642af36ce7d738ab6058b063ec58e0859
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * TabBar
  * ExtremeProDark 2.3.3
  * FreeCAD-BlenderBIM 2022.10.20
```

Closing.

closed 2022-11-20 by luzpaz