FreeCAD/FreeCAD #7853: [Problem] Groove in pipe from padded concentric circles breaks geometry of pockets

FalcoGer, 2022-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Ubuntu 22.04 LTS (`Linux [host] 5.15.0-53-generic #59-Ubuntu SMP Mon Oct 17 18:53:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`)
Version 0.20.1, built 2022-10-27 T 15:41:00
Hash: 22f524c
```

### Subproject(s) affected?

PartDesign

### Problem description

When padding a pipe from concentric circles, then adding holes via pocket and polar patterns to the pipe and then adding a slot with the grove tool inside those circles breaks the geometry of the pockets. I have seen pockets becoming triangular and also wild geometry artifacts like dislodged polygons sticking into the pipe.

Doing the very same thing from a pipe created by revolving a rectangle creates the shape without issues.

### Anything else?

These two pipes are exactly the same, except one was created with padding, the other by revolving.

[image]

Updating the geometry may fix the issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FalcoGer, 2022-11-21:
I tried uploading the example file, but github doesn't want it. Here is the exact steps to reproduce the issue.

1. Open Part Designer
2. Create a body
3. Start a new sketch on the XZ-Plane
4. Create two circles with the center on the origin, 50mm, 55mm
5. Pad the sketch to 150mm
6. Create a datum plane with reference to the XY-Plane and offset it by 55mm (r2 of the pipe)
7. Create a sketch on the datum plane
8. Add a circle on the Y axis, 50mm from the origin and 20mm in diameter
9. Add a pocket to the sketch, "up to first" to create a hole in the pipe
10. Add a polar pattern to the pocket, 45°
11. Create a new sketch on the YZ-Plane
12. Draw a rectangle, height is 5mm (wall thickness of the pipe (55mm-50mm)), width is the hole diameter (20mm), horizontal offset of the lower corner is 40mm (circle distance - diameter / 2), vertical offset is 50mm (pipe inner radius)
13. Create a grove on the rectangle, angle is 45° (polar pattern angle), reference axis is the Y axis
14. Observe broken geometry

luzpaz, 2022-11-21:
Please paste full [About](https://wiki.freecad.org/About) info in original ticket above

luzpaz, 2022-11-21:
Have you asked anyone on the forum to confirm this before opening a ticket?

FalcoGer, 2022-11-22:
```
OS: Ubuntu 22.04 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: C/Default (C) [ OS: English/United States (en_US)
```

I don't know about any forums and I don't want to register with yet another website I will only use once. Why would I need to confirm anything? It's a bug, it happens. I added the steps to reproduce and a screenshot. What's to confirm?

luzpaz, 2022-11-23:
We recently changed our approach to bug tracking. We used to have bug reporters open a [forum](https://forum.freecadweb.org) thread to confirm the bug was actually a bug (by having a tester test on the latest dev version. Anyway, is it possible for you to upload an example file ?

FalcoGer, 2022-11-25:
Here is the file.
https://drive.google.com/file/d/1GOXRaAf-rtFMWnr2jpm9kXLv62TinVm7/view?usp=share_link
I might take it down later, probably in a few months or so. Github doesn't accept it, even though they say zip files are supported, I can't upload here.

adrianinsaval, 2022-11-29:
looks like an occt coplanar issue and it seems to be fixed in 7.6 (maybe by the blobfish patches), here I have fully recomputed every element of that body and it now works, I also upload the original for reference in case the google drive file is later removed
[PipeWithGrove.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10113635/PipeWithGrove.FCStd.zip)
[PipeWithGrove-recomputed.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10113637/PipeWithGrove-recomputed.FCStd.zip)
(zip extension is a dummy, just rename)

> OS: Windows 10 Version 2009
> Word size of FreeCAD: 64-bit
> Version: 0.21.31097 (Git)
> Build type: Release
> Branch: master
> Hash: ccf074008774f1bdfed8fec09ed60e2c0fe77cae
> Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
> Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
> Installed mods:

closed 2022-11-29 by luzpaz

adrianinsaval, 2022-11-29:
It's pointless, ubuntu does not update these things, only on new releases they pull whatever debian has. At most the freecad ppa version could be updated but I think FalcoGer is using the one from the ubuntu repos.

luzpaz, 2022-11-29:
Yea, they should just remove FC from the Ubuntu downloads. It's a time-suck for everyone...especially the users who are trying to learn FC and get a very bad impression because of packaging issues.

FalcoGer, 2022-11-30:
> It's pointless, ubuntu does not update these things, only on new releases they pull whatever debian has. At most the freecad ppa version could be updated but I think FalcoGer is using the one from the ubuntu repos.

That's not correct. I use the freecad ppa at https://ppa.launchpadcontent.net/freecad-maintainers/freecad-stable/ubuntu/ for jammy.

adrianinsaval, 2022-11-30:
> That's not correct. I use the freecad ppa at https://ppa.launchpadcontent.net/freecad-maintainers/freecad-stable/ubuntu/ for jammy.

Ok I was confused because the ppa has opencascade 7.5.2 and your version info says 7.5.1, but you're right it seems there isn't a package for ubuntu 22.04 (btw please use numbers, these version names are meaningless to most people that aren't actively using ubuntu).

> I also don't feel like updating all my software manually by downloading releases from git or compiling myself every week.

well we have snap and flatpak packages too, both using occt 7.6, these are very easily updated. And the appimage (which also uses occt 7.6) can also be easily updated using [AppImageUpdate](https://github.com/AppImageCommunity/AppImageUpdate/releases/latest), it only needs to download the difference, not the whole appimage. I understand the appeal of the ppa but sadly it's not in the best state right now.

@kkremitzki is it possible to update the version of occt in the ppa? current version is buggy, other official releases of FreeCAD are using 7.6.3 now.

FalcoGer, 2022-11-30:
> these version names are meaningless to most people that aren't actively using ubuntu

I used the name jammy because that's what the directory in the ppa is actually called and that is missing from the weird dependency repository. I don't know about any of the versions of dependencies. I simply added the ppa to my sources and ran `apt install freecad`, and that is what I got.

adrianinsaval, 2022-11-30:
I understand why you would use the name, I am just recommending to use numbers in the future