FreeCAD/FreeCAD #7856: [Issue] Segfault when opening task on Detail View for which the parent View was deleted

FalcoGer, 2022-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 22.04 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
[/code]
```

### Issue description

When opening the tasks on a detailed view for which the parent view was deleted, the program crashes with a segmentation fault.

Detailed steps to reproduce:

1. Create a new document
2. Open Part Designer Workbench
3. Create a new body
4. Create Cube Primitive with default values
5. Open Technical Drawing workbench
6. Create a new default page
7. Create a view with projection view and add the bottom view to the default view
9. Create a detailed view on the bottom view
10. Double click the projection view and remove the bottom view by unchecking the checkbox
11. Double click the detailed view in the tree
12. Click OK
13. Segfault

### Anything else?

Here's the stacktrace:
```text
TaskDetail - no BaseView.  Can not proceed.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f367de1b520]
#1  0x7f367f3214fc in QGraphicsItem::setVisible(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xc
#2  0x7f35777d0c55 in TechDrawGui::TaskDetail::accept() from /usr/lib/freecad-python3/lib/TechDrawGui.so+0x45
#3  0x7f35777cdb65 in TechDrawGui::TaskDlgDetail::accept() from /usr/lib/freecad-python3/lib/TechDrawGui.so+0x15
#4  0x7f36807d9d4a in Gui::TaskView::TaskView::accept() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x6a
#5  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f17b8) [0x7f367e5427b8]
#6  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x30e670) [0x7f367f19f670]
#7  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f17b8) [0x7f367e5427b8]
#8  0x7f367f0f3116 in QAbstractButton::clicked(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#9  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26239e) [0x7f367f0f339e]
#10  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x263fc4) [0x7f367f0f4fc4]
#11  0x7f367f0f51e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#12  0x7f367f0404ee in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#13  0x7f367effd713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#14  0x7f367f005364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#15  0x7f368056ba10 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#16  0x7f367e50ae2a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#17  0x7f367f003e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#18  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7f367f059d40]
#19  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7f367f05cfd5]
#20  0x7f367effd713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#21  0x7f368056ba10 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#22  0x7f367e50ae2a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#23  0x7f367e8f2307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#24  0x7f367e8c7a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#25  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7f3677fa4d6e]
#26  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f367c0b6d1b]
#27  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xaa6f8) [0x7f367c10b6f8]
#28  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f367c0b43c3]
#29  0x7f367e5640a8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#30  0x7f367e50974b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#31  0x7f367e511ce4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#32  0x7f3680501162 in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x2012
#33  freecad(+0x50cb) [0x56310f4c20cb]
#34  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f367de02d90]
#35  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f367de02e40]
#36  freecad(+0x60d5) [0x56310f4c30d5]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-11-23:
fixed in master by cdda6f2c38.

Still working on backport to v020

donovaly, 2022-11-29:
> Still working on backport to v020

Thanks for the backport.
Fixed for FreeCAD 0.20.2 by #7898

PR #7898 (2022-11-29): [TD]backport fix for no Detail without BaseView
Fixes #7856 in v0.20.

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [ ]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the feature [... cut]
closed 2022-11-29 by donovaly