FreeCAD/FreeCAD #7860: Sketcher: Dimensions get unreadable (small) if a 2nd sketch is visible

purehunter1, 2022-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=13&t=43670

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes
  * Curves
  * fasteners
  * Lithophane
  * sheetmetal
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

If you have 2 sketches in your project e.g. a profile (1st) and a path (2nd), perpendicular to the profile, the dimensions of the opened sketch (1st, profile) can get unreadable small if you zoom into your sketch.

I made 4 important observations:
1. only the dimensions are affected. Other symbols for constraints are not.
2. It is independent of the chosen font size for the sketcher.
3. The problem only occurs, if the 2nd sketch is "visible". If you toggle the visibility of the 2nd sketch in the model path while working on the 1st sketch, you can switch on and off the problem (dimensions readable or not)
4. It is obvious, that the severity of the shrinking of the dimensioning depends on the geometry of the 2nd sketch. If there is a point in the 2nd sketch which is virtually behind the viewer, the severity of the shrinking is highest. If the path is very long, e.g. 5000mm, one point of it is always behind the viewer and working on the profile with the path visible is impossible.

### Anything else?

This problem already existed in FC 0.18.1 or even earlier.
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-23:
Please add an example file

purehunter1, 2022-11-23:
[ZoomTest.zip](https://github.com/FreeCAD/FreeCAD/files/10074405/ZoomTest.zip)
I zipped it, as FC files are not accepted

adrianinsaval, 2022-11-23:
This is a duplicate of #7082 for me having the other sketch visible or not makes no difference, but changing between perspective and parallel view does. To work around the issue go to sketcher settings and enable the option to force parallel view when entering sketcher edit mode

commit 8a0816ff (2022-11-24): Sketch: use the focal distance to determine a correct scale factor for SoDatumLabel

This fixes #7860: Dimensions get unreadable (small) if a 2nd sketch is visible
and fixes #7082: Dimension size on perspective view scales "backwards"
commit 2605b54d (2022-11-24): Sketch: use the focal distance to determine a correct scale factor for SoDatumLabel

This fixes #7860: Dimensions get unreadable (small) if a 2nd sketch is visible
and fixes #7082: Dimension size on perspective view scales "backwards"
closed 2022-11-24 by wwmayer