FreeCAD/FreeCAD #7872: [PD]: Pocket up to face/up to first creates traces and multiple faces (refining doesn't work)

69Piero, 2022-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=19&t=73809  https://forum.freecadweb.org/viewtopic.php?f=8&t=73760

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.21.31058 (Git)
Build type: Release
Branch: master
Hash: e6c53d8642af36ce7d738ab6058b063ec58e0859
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.54b
  * Assembly3 0.11.4
  * Assembly4 0.11.8
  * Behave-Dark-Colors 0.1.1
  * BOLTSFC
  * CurvedShapes 1.0.4
  * Curves 0.6.1
  * dxf_library
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fasteners
  * FCGear
  * freecad.gears 1.0.0
  * ImportNURBS
  * InventorLoader 1.3.0
  * InventorLoader.backup1663313199.3655012
  * invgears 0.1.2
  * lattice2
  * nurbs
  * Render 2022.1.0
  * sheetmetal 0.2.59
  * sheetmetal.backup1663313192.0420823
  * Silk -1321532872.0.0
  * symbols_library
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

Refining (forced and auto when set in preference) doesn't work in pocket up to face/up to first. 
Eventual following features wake up the refining but if for some reasons the cut is compulsory the last feature in the model we get an incorrect outcome with traces and multiple faces. The error spreads also to exported step for example

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-11-24:
please attach a file where the issue can be reproduced, you can rename the file to have .zip extension so github accepts it

69Piero, 2022-11-24:
that an example

[Bug Up to face-up to first.zip](https://github.com/FreeCAD/FreeCAD/files/10084952/Bug.Up.to.face-up.to.first.zip)

luzpaz, 2022-12-04:
@FlachyJoe would you mind taking a look at this bug? (in lieu of #7880)

grubertechnic, 2023-07-04:
Still an issue now.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33487 (Git)
Build type: Release
Branch: master
Hash: 3cb5bdcbe4266b933b3c7a53055cc4212cd3cfa1
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

tarman3, 2024-12-20:
Can not reproduce

[image]

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

closed 2025-06-09 by github-actions[bot]