FreeCAD/FreeCAD #7873: [Problem] A2Plus does not recognize user units preferences in constraint dialog

hl1234, 2022-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.31058 (Git)
Build type: Release
Branch: master
Hash: e6c53d8642af36ce7d738ab6058b063ec58e0859
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60e
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

I set my preferences to Building US. This is reflected in standard workbenches (at least Part Design and Sketcher).  However when I use A2Plus for assembly and select the coincident constraint a) no units show up in the spin box and I am unable to set them. The numbers entered are definitely interpreted as mm. The decimal points are retrieved correctly and change based on my general FreeCad Units Preferences. I have been away from FreeCad and A2Plus for a year or two and I am pretty sure this used to work. It looks like there are a couple of things going on here. My testing has been in a2p_constraintDialog.py

1. It looks like the method to retrieve the units is different from that for decimal points. 

<pre>
FreeCAD.Units.Quantity(1,FreeCAD.Units.Length))[3:]
vs.
FreeCAD.ParamGet("User parameter:BaseApp/Preferences/Units")
</pre>

I have not done any development so perhaps there is a reason for this, however the method for units returns 'mm'. So there is an attempt to retrieve the units but an incorrect value is returned.  I inserted an ConsoleMessage to show this.
 <pre>
FreeCAD.Console.PrintMessage("Found units >{}<\n".format(str(FreeCAD.Units.Quantity(1, FreeCAD.Units.Length))))
  11:19:35  Found units >1.0 mm<
</pre>

2. The QtGui.QDoubleSpinBox likely only accepts numbers so does not provide the ability to insert a unit override.  It appears that the code is looking for an override but maybe that is not intended functionality. Looking at GitHub it seems like the spin box was introduced at be14bde1f7ce2f4880cae314d4ff9766502cb038 Dec 7 2019 which might not be an issue if the idea of abandoning the ability to enter units override at input is okay, and obviously fixing how the units are retrieved. I haven't looked at the docs for QtGui.QDoubleSpinBox in detail so maybe it can support the spinning digits and the units.

3. I was able to override the 'mm' fetched as described above by inserting
<pre>
    user_unit = '"'
</pre>
When I do this a " does show up in the spin box so maybe that is a legit usage however in the unmodified version mm does not display in the spin box.

 I have begun looking for documentation / threads that describe where to retrieve user preferences by Python.

### Anything else?

<pre>
index 2a45883..60f9d29 100644
--- a/a2p_constraintDialog.py
+++ b/a2p_constraintDialog.py
@@ -146,6 +146,9 @@ class a2p_ConstraintValueWidget(QtGui.QWidget):

             # get the length unit as string
             self.offsetEdit.setSuffix(" " + str(FreeCAD.Units.Quantity(1, FreeCAD.Units.Length))[3:])
+            # hl1234: debug messages
+            FreeCAD.Console.PrintMessage("Found units >{}<\n".format(str(FreeCAD.Units.Quantity(1, FreeCAD.Units.Length))))
+
             # the maximum is by default 99.99 and we can allow more
             self.offsetEdit.setMaximum(1e7) # allow up to 1 km
             # set minimum to negative of maximum
@@ -178,6 +181,8 @@ class a2p_ConstraintValueWidget(QtGui.QWidget):
                 else:
                     user_unit += c

+            # hl1234: Jam inches in here
+            user_unit = '"'
             user_qty = float(user_qty.replace(",","."))

             #self.offsetEdit.setValue(offs.Value)
</pre>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hl1234, 2022-11-25:
Okay. After a little more sensible testing I tried a few of the other non-metric Units Systems in preferences and all is well for those. So there is something much more subtle going on specifically with Building US.

luzpaz, 2022-11-29:
Please open this ticket in the A2plus repo https://github.com/kbwbe/A2plus

closed 2022-11-29 by luzpaz