FreeCAD/FreeCAD #7875: [Feature Request] Bigger navicube corners

adrianinsaval, 2022-11-24:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=34&t=68544

### Subproject(s) affected?

Core

### Idea description

Currently navi cube corners are small and can be hard to click, one proposed solution is to use a bigger hexagonal shape instead of the current triangular shape
[image]
[image]

here the lines of code that likely need to be changed:
https://github.com/FreeCAD/FreeCAD/blob/a6b9a53bed2775107d8b8d21eedab5cdba3fdf2f/src/Gui/NaviCube.cpp#L422-L438

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

silopolis, 2022-11-24:
YES! 😍
Please 🙏

MisterMakerNL, 2022-11-24:
IT's cool, not original but it's better.

adrianinsaval, 2022-11-24:
almost there:
[image]
I don't understand well how the transformations of the corner work so I don't know how to properly adjust the position, also I don't know why a square is still being drawn on the sides on top of the filleted squares

adrianinsaval, 2022-11-25:
thanks to the help from @Jolbas it's almost ready now:
[image]

adrianinsaval, 2022-11-25:
made a PR #7876

commit dfaecfa1 (2022-11-29): [fixes #7875] Gui: use hexagon shape in NaviCube corners

makes the corners bigger and therefore easier to click

fix inconsistent NaviCube borders
closed 2022-11-29 by donovaly